carla-simulator/carla #1473: why doesn't the jstest-gtk show the Logitech G920 Driving Force Racing Wheel?

0102TH, 2019-04-01:
On Ubuntu 16.04,when I inserted the usb of the wheel into my computer,opened jstest-gtk,but nothing showed.in the terminal,input the 'lsusb' or 'dmesg'.all things are normal.I'm so confused about that..
therefore,I am here for any valuable advices,thanks in advance.

this is the output after the 'dmesg'
[  784.855156] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[  785.004781] usb 1-2: New USB device found, idVendor=046d, idProduct=c261
[  785.004787] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  785.004792] usb 1-2: Product: G920 Driving Force Racing Wheel for Xbox One
[  785.004796] usb 1-2: Manufacturer: Logitech
[  785.004800] usb 1-2: SerialNumber: 0000b450139a7734

sangeetsu, 2019-04-16:
Hey,
Ubuntu 16.04 doesn't have support for G920 natively. Although, you can add support using usb-modeswitch. You can look at these:
http://burning-midnight.blogspot.com/2017/11/logitech-g920-steering-wheel-on-linux.html

http://www.draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=2&t=2510

0102TH, 2019-04-17:
@sangeetsu 
Thank u for the sharing.I will try it again.
I have another solution: when it does not display the steering wheel, I restart the computer to the Windows system, run the official driver software of Logitech game software, modify the parameters of the steering wheel, and then change the parameters back. Then open the BeamNG game and let the steering wheel correctly perform the parameter changes I have done before. After that, restart the computer to the Ubuntu system. At this time, it displays the steering wheel. Although the problem is solved, I still don't know why.
when jstest-gtk does not show the steering wheel,I enter lsusb in the terminal,terminal shows:
ID 046D:c261 Logitech,Inc
when jstest-gtk display the steering wheel, I enter lsusb in the terminal，terminal shows：
ID 046D:c262 Logitech,Inc

sangeetsu, 2019-04-17:
Its because the steering is in xbox mode by default. It needs to use usb-modeswitch to bring it to something a pc can interact with. Hope that makes sense.

0102TH, 2019-04-17:
@sangeetsu 
That's it.
Thank u, I would try it later.

closed 2019-06-23 by stale[bot]