carla-simulator/carla #1475: CARLA scripts IndexError: list index out of range

Iswed, 2019-04-01:
CARLA 0.9.4 running on Windows 10. There seems to be an issue with the automatic_control and manual_control scripts when running on python 3.7 with pygame 1.9.5. The error occurs in the initialisation of the HUD instance. An indexing error in the list of fonts relating to the following lines:
fonts = [x for x in pygame.font.get_fonts() if 'mono' in x]
mono = default_font if default_font in fonts else fonts[0]

the font 'mono' is not found in pygame.font.get_fonts(), therefore the list 'fonts' is empty. Calling fonts[0] results in the indexing error.
Replacing 'mono' with another font such as 'arial' or another font in pygame.font.get_fonts() seems to resolve the issue, as fonts[0] will not be null.

rajcivils, 2019-12-21:
Just replace the mono with arial. It works for me. This was told in sentdex series. And this bug can be solved by just replacing all monos with arial. Which is super simple in sublime. 
[image]

Just click on replace all. In mine it has already been done. But note it is actually mono there if you are getting the error.

closed 2020-03-19 by germanros1987

lshmouse, 2022-09-26:
See: https://github.com/pygame/pygame/issues/3156 

In uubuntu,  the reason why pygame.font.get_fonts() returns none is 'fc-list' not found, the following command may fix it.

```
sudo apt install fontconfig
```