carla-simulator/carla #1476: Issue creating .egg file from Carla repository

SIlvaMFPedro, 2019-04-01:
Good afternoon. I am an engineering student and i am using CARLA for my master's thesis.
I am having an issue creating .egg file from the CARLA repository. 
[image]
It says it cannot find Memory.h file but this file exists in the "~/carla/Libcarla/source/carla" directory.

Can someone help me fix this issue?

Thank you very much.

Pedro Silva - Aveiro University - DETI

nsubiron, 2019-04-01:
Hi @SIlvaMFPedro,

Use `make PythonAPI` on the root folder.

SIlvaMFPedro, 2019-04-01:
Already solved the issue @nsubiron .
Thank you for the help ;)

closed 2019-05-31 by SIlvaMFPedro