carla-simulator/carla #1478: failed to read data: timed out when run driving_benchmark_example.py --corl-2017

AhmdNassar, 2019-04-02:
I use carla 0.8.2 when I run manual_control.py from PythonClient it worked fine, but when I want to run driving_benchmark_example.py --corl-2017 I get error : 

ERROR: (localhost:2000) failed to read data: timed out
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host

and repeat it self till I close cmd

Johnnyswong, 2019-04-02:
Ah, this bug is related to https://github.com/carla-simulator/carla/issues/697. This rpclib bug exists in the carla version before 0.9.3.

marcgpuig, 2019-04-05:
Hi @AhmdNassar 
Has @Johnnyswong answer solved your question?

AhmdNassar, 2019-04-05:
> Ah, this bug is related to #697. This rpclib bug exists in the carla version before 0.9.3.

thank you for your replay, but in #697 he had a problem with client_example.py but it worked fine with me, I have a problem with driving_benchmark_example.py although they used the same port!

closed 2019-06-11 by stale[bot]