carla-simulator/carla #1480: How to configure LIDAR sensor for point cloud registration in Carla 0.9.4

kartik144, 2019-04-02:
> The files under "Deprecated" folder are not compatible with the last version, we're just keeping them until we port everything to the new API. You should see the examples in "PythonAPI".
> 
> In the new API you can adjust the frequency in the lidar blueprint
> 
> ```python
> bp = blueprint_library.find('sensor.lidar.ray_cast')
> bp.set_attribute('rotation_frequency', '10')
> ```

Hi, can anyone tell what all attributes are present in the LIDAR sensor blueprint? I want to capture LIDAR in a frame similar to other sensors, i.e. through a dashboard camera for point cloud registration between the LIDAR points and semantic segmentation images, similar to what has been done in [this](https://arxiv.org/pdf/1809.11036.pdf) paper on page 11, Figure 4. 

Can someone tell me how do I have to configure my LIDAR sensor for that?

_Originally posted by @kartik144 in https://github.com/carla-simulator/carla/issues/995#issuecomment-478931793_

Johnnyswong, 2019-04-02:
All attributes for the LIDAR sensor are listed in https://carla.readthedocs.io/en/latest/cameras_and_sensors/ .

kartik144, 2019-04-03:
Might have missed that. Thanks a lot for the help!

closed 2019-04-03 by kartik144