carla-simulator/carla #1486: Lane type info issue

akki6843, 2019-04-03:
Hello,
I am trying to access the lane_type info for the lane which is on right hand side of the 
ego_vehicle.  

And I am getting following results:
* current lane is driving 
* first right lane is parking 
* second right lane is sidewalk   

But if we look at **below image there is no parking lane.** Is this expected?

        _current_lane_info = self.ego_vehicle.get_world().get_map().get_waypoint(self.ego_vehicle.get_location())
        print(_current_lane_info.lane_type)
        if(str(_current_lane_info.lane_type) == 'Driving'):
            _new_lane = _current_lane_info.get_right_lane()
            while(True):
                print(self._num_lane_changes, _new_lane.lane_type)
                if((str(_new_lane.lane_type) == 'Driving') or (str(_new_lane.lane_type) == 'Parking')):
                    _new_lane = _new_lane.get_right_lane()
                    self._num_lane_changes += 1
                else:
                    break
        else:
            _new_lane = _current_lane_info

[image]

XGodina, 2019-04-04:
Hi @akki6843, 
There is an unmarked parking lane there. We working with that.

closed 2019-06-10 by stale[bot]

noahcao, 2019-08-16:
Hi @XGodina ,
I think there is some issue for maps on 0.9.4? No matter where the vehicle is, the lane type attached to its waypoint is always "driving". I checked the opendrive .xodr file, I think the lane type definition inside is right. So I believe there is something wrong with the lane id annotation for the carla map.

MelihCelik00, 2021-08-09:
I know this was an old issue but is there a chance that it is still not fixed? I still also get 'driving' on all of my map which is my custom-made map. Maybe it was my map's problem but I wondered if the problem is still not fixed

XGodina, 2021-08-16:
Hi @noahcao, ok you mean there is some lane information that doesn't represent by mesh or texture on the map. Right ? I will take a look. Thanks