carla-simulator/carla #1487: Vehicle creation on a particular lane on particular position

Narendarselva, 2019-04-03:
Hi Team,
I m working on a project which requires creation of vehicle on timely interval on a desired location on a desired lane . Is it possible using crala simulator ??
I gone through the PythonAPI documents there are lot of options which can use random spawn_points which choices randomly . And i dont know exactly how the lane based APIs are used .
Kindly help me on this .
Is there any specific examples for this is also welcome .
Thanks 
Naren

nsubiron, 2019-04-04:
Hi @Narendarselva,

You can spawn vehicles at any location/rotation in the map, the only problem is that the autopilot won't work correctly everywhere, that's why we have recommended spawn points.
```py
transform = carla.Transform(carla.Location(x=1,y=2), carla.Rotation(yaw=90))
actor = world.spawn_actor(blueprint, transform)
```

Alternatively, you can use waypoints to project locations to the road
```py
waypoint = map.get_waypoint(carla.Location(x=1, y=2))
actor = world.spawn_actor(blueprint, waypoint.transform)
```

Narendarselva, 2019-04-05:
Thanks a lot.
1. But is there any provision to like toolkit to get the location (x,y,z) on the map based on mouse pointer movement. So that we can get the correct coordinates (x,y) where a vehicle can be spawned using that .As of now i see no toolklit , just a pygame window with HUD informations.
2. Is there a way we can convert the x,y,z locations of any actors(like speedlimit board or traffic lights) into respective LAT/LONG ??

nsubiron, 2019-04-05:
1. That would be indeed a cool feature, but we don't have such thing. For now you can drive with manual_control.py to the location you want and annotate the vehicle's location.

2. We added in last release (0.9.5) the `transform_to_geolocation` function to the map, it transforms a carla.Location to a carla.GeoLocation (latitude, longitude, altitude).

Narendarselva, 2019-04-08:
Thanks a lot.
Is the transform_to_geolocation is provided as PythonAPI ??
How i can make my route in a MAP ??
I could see predefined routes and spaw_points inside all maps , But what if i want to create my own route in MAP in UE4 editor?

Derekabc, 2019-04-19:
Hi @Narendarselva , I have a similar problem as you,  I want to use the multi-lane senario of the Town3. So I want to put some car models along with a self-driving car on the particular position, Have you figure out how to implement this by a simple way?

closed 2019-06-25 by stale[bot]

talsperre, 2019-06-26:
Hi @Derekabc, could you find any way of positioning car models in other lanes near the ego vehicle?

Derekabc, 2019-06-26:
you can use the code like this : 
```
 transform = world.player.get_transform()
nearest_waypoint = world.map.get_waypoint(transform.location, project_to_road=True)
left_wp = nearest_waypoint.get_left_lane()
right_wp = nearest_waypoint.get_right_lane()
```
to inform the waypoint of the lane next to the ego car and then spawn a car on that point.

talsperre, 2019-06-26:
@Derekabc , thanks a lot. This works perfectly for my use case. Although, I had to generate the waypoints approximately 5-10m ahead in order to avoid collision.

lichunly, 2020-05-22:
@Derekabc hi, the function that getting the location on map by mouse-pick is possible now(carla 0.9.9)?