carla-simulator/carla #1488: How to set traget/max speed (5kmph) for auto pilot?

AravindChandradoss, 2019-04-03:
Is there any way to set the target/max speed for Adaptive Cruise Control (ACC) in Autopilot mode. 
I could not find such settings in _manual_control.py _or_ automatic_control.py_. I will be glad to know the location for ACC code.  

I think, by default, it was set to 20kmph. I need to set it like 5kmph. 

Btw, I am trying to use LIDAR for mapping the environment. So, I need a slow moving car.

Thanks!

AravindChandradoss, 2019-04-03:
For now, I am using automatic_control.py with few changes in basic_agent.py. 
`def __init__(self, vehicle, target_speed=20):` to `def __init__(self, vehicle, target_speed=5):`

It is working!
 
But I will be glad if anyone tell how to implement the same for manual_control.py (Coz, autopilot mode in manual_control.py is smooth and better)

nsubiron, 2019-04-04:
Hi @AravindChandradoss,

The autopilot in manual_control.py is run on server-side, the code is in [WheeledVehicleAIController.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp).

nsubiron, 2019-05-08:
Hi @bigsheep2018,

Yes, "spawn_npc.py" uses the autopilot in server-side (UE4).

Guanwh, 2019-05-24:
@AravindChandradoss I also want to set to 5km/h. Can you tell me how to do it? 
Thanks in advance.

AravindChandradoss, 2019-05-28:
@Guanwh  check these lines 483-495 in [this code](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp)

Sorry for late reply.

Guanwh, 2019-05-31:
@AravindChandradoss  Thanks for your reply! But I had tried it, it didn't work for the auto pilot!

AravindChandradoss, 2019-05-31:
you have to recompile. do `make launch` in your carla directory. refer [this](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/) if needed.

closed 2019-05-31 by AravindChandradoss

kannidekan, 2020-04-24:
Check out the **Traffic Manager** in Python API.

>  The vehicles will drive at 80% their current speed limit, leaving at least 5 meters between them and never perform a lane change.
`tm = client.get_trafficmanager(port)
tm_port = tm.get_port()
for v in my_vehicles:
  v.set_autopilot(True,tm_port)
danger_car = my_vehicles[0]
tm.global_distance_to_leading_vehicle(5)
tm.global_percentage_speed_difference(80)
for v in my_vehicles: 
  tm.auto_lane_change(v,False)`

[here are the link](https://carla.readthedocs.io/en/latest/adv_traffic_manager/)

snehanjalikalamkar, 2023-02-08:
> @Guanwh check these lines 483-495 in [this code](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp)
> 
> Sorry for late reply.

The code is 190 lines, could you please tell me where exactly/which variables in the CPP file must be changed?