carla-simulator/carla #1493: Find occlusion object in Image plane

dldudwo0805, 2019-04-04:
Hi, I am drawing 2d-bounding box of objects in the image plane.
[Camera]([url](url)) is located in ego-vehicle, far from (1.6, 0, 2) **in** the center of the vehicle.
I use the object's global position, and then projection to the image plane.
Although object in front of the camera, but occluded by building or others, that object is drawn.
How can I fixed that situation. I want draw 2d-bounding box which is not occluded.

[image]
[image]

baurst, 2019-04-04:
This is probably not the best way to do it, but you could use an additional depth camera with the same parameters as the rgb camera and check if the objects distance to your ego vehicle is larger than the distance you get from the distance image.

dldudwo0805, 2019-04-04:
Thank you for giving idea.
Now I turn into using segmentation data.
Using that data, delete the occlusion data, similarly that you mentioned.

marcgpuig, 2019-04-05:
I would have went for the depth option too :)
Closing the issue since seems solved!

closed 2019-04-05 by marcgpuig

Vin1291, 2020-07-02:
@dldudwo0805 Hello, I am stuck on the same thing, can you share how you solved the issue using semantic segmentation.

rohansingh080699, 2023-03-31:
@marcgpuig @baurst , Hello! I am trying to do the same thing as stated in this issue but in carla 9.13  and carla 9.13 has an instance segmentation camera. So basically I have two small questions, is the instance segmentation camera better or the depth camera better  to remove the 2d bounding boxes of occluded objects?(can you also please tell me how to approach this problem?)

Coming to instance segmentation camera, i have set up an instance segmentation camera with the same parameter as the RGB Camera and I'm also listening to the output and saving it in a dictionary and I know that the tag information encoded in the red channel, but my doubt is how do i filter out the occludes 2d bounding boxes with this tag information?

rohansingh080699, 2023-04-01:
Hello @dldudwo0805 and @Vin1291 , were you guys able to find a solution? if yes, can you please help me out with this issue?