carla-simulator/carla #1494: Running carla in Docker

Johnnyswong, 2019-04-04:
Hi, @nsubiron 
I try to run the carla docker container. But failed with the error:
`Failed to find symbol file, expected location:
"/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4.sym"`

My gpu card is a P40 and a V100. Their gpu driver version are both 390.87.
I can start the carla docker container successfully in my local pc, with a 1080ti gpu card and gpu driver version 390.87. 

Previous issues about this problem have pointed out that it may related to the OpenGL library. I have installed the OpenGL library on my server, but it didn't work still.

After some googling, I think the problem is related with the Unreal4.21, the following link is the similar problem:
`https://answers.unrealengine.com/questions/886214/view.html`.
But I have no idea how to fix it.

rkollataj, 2019-05-20:
I had similar problem (also running from docker) in CARLA 0.9.5. My GPU: Radeon HD 8890M. Running in low quality helped in my case:
`./CarlaUE4.sh -quality-level=Low`
After the above I can run Carla without a problem even in Epic quality.

Johnnyswong, 2019-06-13:
> I had similar problem (also running from docker) in CARLA 0.9.5. My GPU: Radeon HD 8890M. Running in low quality helped in my case:
> `./CarlaUE4.sh -quality-level=Low`
> After the above I can run Carla without a problem even in Epic quality.

Thanks for your reply. I tried your method but it still can't work yet.

qhaas, 2019-07-24:
I mysteriously started getting this in Ubuntu 18.04 when running via Singularity 3.0 and then mysteriously stopped getting it.  Might be a coincidence, but when I was getting this error, there was another CUDA process (Type C) sharing GPU 0 (which CARLA seems to be hardwired to use) observed using `nvidia-smi -i 0`).  I fear t his error has more than one cause.

Kailiangdong, 2019-07-24:
@Johnnyswong Hello, where do you find the support information of OpenGL about your GPU? I can't find what version support my tesla p100, thank you

Kailiangdong, 2019-07-30:
@Johnnyswong Do you finally run carla on your tesla server? Thank you

closed 2019-10-05 by stale[bot]

Jiankai-Sun, 2019-12-17:
Same problem for `carlasim/carla:0.9.5`, when I run `docker run --net=host --rm -d -it -e NVIDIA_VISIBLE_DEVICES=0  --runtime nvidia carlasim/carla:0.9.5'
        carla_arg = ' ./CarlaUE4.sh /Game/Carla/Maps/Town03 -benchmark -carla-server -fps=20 -world-port=%s -windowed -ResX=4 -ResY=4 -carla-no-hud`.

The docker container can just spawn some objects but freezed after that.

[image]