carla-simulator/carla #1495: How to add objects to the existing map?

c1enet, 2019-04-04:
Hello, 

I have only a little understanding of unreal engine and I've been trying to modify the existing map (Town03), deleting some of the objects and buildings and adding new ones using UE editor. 

After adding objects to desired location and saving, it is shown on editor's viewport but not in carla world when I launch the server.
Is there anything I am missing? 

Thank you for your time!

marcgpuig, 2019-04-05:
Hi @c1enet 
If you hit the play button you will be able to connect your python scripts there, with your new modified scene. If you want you can make a package (using `make package` on Linux), that will create a build of the project withh your current changes on the editor.
Hope this helps

c1enet, 2019-04-05:
@marcgpuig Thank you for your reply. 
I think that's what I did but it didn't work. Somehow updating client using ./Update.sh sorted it out. 
But now, I am having another issue. When I record the Lidar data (ply). It only shows the half of the radius in one file, which didn't happen in 0.9.3. 
Any idea?
Thank you!

closed 2019-06-11 by stale[bot]