carla-simulator/carla #1500: Get vehicle dynamic limits

togaen, 2019-04-05:
I would like to be able to get nominal vehicle limits for things like: turning radius, top speed, max longitudinal accel/decel, etc.

I know some can be tricky to compute exactly due to dependencies on state, but even just having nominal values would help a lot. How can I get these values with the PythonAPI?

nsubiron, 2019-04-08:
Hi @togaen,

Some of these values are included in the [VehiclePhysicsControl](https://carla.readthedocs.io/en/latest/python_api/#carlavehiclephysicscontrol) (use `vehicle.get_physics_control()`).

togaen, 2019-04-08:
@nsubiron I don't see any of those values there; can you tell me which members they are?

nsubiron, 2019-04-08:
Ok you are right, not directly included, cause most of them are not really known AFAIK. The turning radius is related to the `steer_angle` of the front wheels, the top speed depends on a lot of things, but probably `max_rpm` and `mass` are the ones that affect the most. The vehicle dynamics is quite complex with a lot of parameters, [PhysX docs](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html) may give you more hints.

togaen, 2019-04-08:
Yeah, I was sort of afraid that this would be the case. I don't need hyper-realistic physical bounds (at least to start with) I just need to plausible bounds to give my planner something to work with. If I don't provide bounds, my planner might try to command something like 1000m/s^2 acceleration, which is probably not going to work. But it can only know that that won't work if it gets some constraints from the vehicle models.

This kind of information is required to make any guarantees on driving behavior, so I can't imagine other people won't also need it. Are there any plans to make it easy to access?

In the meantime, Is there anything like a set of "nominal" bounds for each vehicle class? Like, the dynamic limits under perfect conditions? The ros_bridge hard codes some values here in `get_vehicle_max_(speed|acceleration|deceleration)`: https://github.com/carla-simulator/ros-bridge/blob/master/carla_ackermann_control/src/carla_ackermann_control/carla_control_physics.py

If those are reasonable, I will just use those.

nsubiron, 2019-04-08:
For the moment we're just interfacing UE4 vehicle's physics (based on PhysX vehicles), but in the future we'd like to support other vehicle models. The values in the ROS bridge seem reasonable for an average car (probably won't work with bikes and trucks, but seem ok for regular cars), I think you can just use those.

togaen, 2019-04-08:
Thanks. I guess the vehicle model itself is irrelevant... the PhysX models must also have bounds, it's just a matter of how to compute them. It would be really useful to expose that in the API somehow.

But I'll use the ros_bridge values for now. And just to clarify, by "reasonable" I meant: are they accurate with respect to the vehicles in UE4? In other words, do you think the vehicles at least as capable as those bounds suggest? Especially the deceleration. Do you think the car models are all capable of 8m/s^2?

nsubiron, 2019-04-08:
I haven't tested but this is relatively easy to measure from the API for a given vehicle.

togaen, 2019-04-08:
Okay, so no one has actually measured this, then? If that's the case, I would like to script some dynamics tests to pre-compute this information. Where do I find information about the maps that are included? Ideally there would be a map like this, but anything with flat, open space would work: https://www.larson.psu.edu/assets/images/maps/Test%20Track%20-%20overhead%20entrance%20road%20labeled%20copy.jpg

closed 2019-06-14 by stale[bot]