carla-simulator/carla #1501: ServerTimeOut 10000 ms regardless my configration

eslambakr, 2019-04-05:
Hello,

I think there are problem with  " **-carla-server-timeout** ", as i am facing this error:
ERROR: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

However, My command as follows:
DISPLAY= ./CarlaUE4.sh /Game/Carla/Maps/Town03 -windowed -ResX=480 -ResY=640 -carla-port=2000 -benchmark -fps=20 -carla-server-timeout=99999999ms -quality-level=Epic

And in my client I set the timeout as follows:
`client = carla.Client(args.host, args.port)`
`client.set_timeout(99999999.0)`

nsubiron, 2019-04-08:
Hi @eslambakr,

Indeed this is a bug, thanks for reporting. This command-line was used for the older versions (0.8.x), but it was never hooked to the new server system.

JMorceaux, 2019-04-26:
Hey guys, 
I am still having this issue, was it fixed in the recent commit ?
Thanks
@nsubiron

nsubiron, 2019-04-26:
No, we haven't fixed it yet.

chankim, 2019-05-02:
Hi, I'm trying to analyze Carla to see how it is implemented.
I set the client.set_timeout(20000.0) but when I stop the server at a break point,
the client is timed out. (I couldn't find where the error is coming from). Other than this 'client.set_tieout()', where should I set timeout? to keep the client work while I am stopping the server?
Carla is a very complex software and I don't know how people are following the work and using it in their research. Isn't there a more friendly document about how it is implemented, I mean one that shows the architecture or overall operation? For example I'm trying to figure out how the camera data is sent to PyGame window. (seems it's not in the protocolbuf data).
Thanks!
@nsubiron

agarwaltanmay, 2019-05-22:
Is there a way to fix the issue with the server timeouts? The server client connection is lost at random which causes problems in running any longer runs.

closed 2019-06-13 by nsubiron

eslambakr, 2019-06-13:
@nsubiron 
Hello,
Excuse me I can't understand how #1741 solves this bug.
I see that you say It isn't related, however you close this bug

Will #1741 solve this bug?
If yes, I will be grateful if you can illustrate more.

Thank you very much

nsubiron, 2019-06-13:
Hi @eslambakr,

It does not solve it, sorry GitHub only has option for closing, I'll mark it as `wontfix`. It turns out rpclib does not support setting a time-out on the server-side, thus we just removed the `-carla-server-timeout` flag as it does nothing.

This problem is probably related to https://github.com/carla-simulator/carla/issues/1125#issuecomment-479781701 rather than the server time-out.

xmyqsh, 2019-06-13:
@eslambakr Sorry. I have tried to reproduce your error report but failed.
Also, I cannot deduce the result as yours from the code.
Could you provide a simpler test case?

eslambakr, 2019-06-16:
@nsubiron @xmyqsh  I read the comment in #1125 and I think it is very related.
I although tried some experiments to find out the problem, Here you are my experiments:
1- I tried to run a simple code but without running the server 
client = carla.Client(host, port)
client.set_timeout(4.0)
and as expected It gives me an time-out error (time-out of 4000ms while waiting for the simulator)

2- But when I tried to run a data collection code which saving a big data while the server is running this time, so the connection is lost while saving on the disk, It gives me time out but regardless my configuration (client.set_timeout(4.0))
It gives me time-out of 10000ms while waiting for the simulator, however I set it to 4!!
why while the server is running it times out in the default configuration not mine?

Thanks for you help

nsubiron, 2019-06-17:
> It gives me time-out of 10000ms while waiting for the simulator, however I set it to 4!!

`world.wait_for_tick(timeout)` has a different timeout, by default 10000ms, it's probably that one.

eslambakr, 2019-06-18:
@nsubiron  Thank you very much, I think it works fine now.

Sorry for the late response as I run the code overnight to make sure the timeout problem is solved, It times out now according to my configuration.

Again thank you for help

eslambakr, 2019-06-26:
Hello @nsubiron 
As I said above the client times out according to my configuration, but however I set it to a big number (ex: one hour) it still time out.

This behavior didn't happen in version 8, so why does it happen in version 9?, however you said that the server tries to reconnect continuously.

What are the possible reasons which cause time-out? to avoid it as possible.

Thanks in advance

nsubiron, 2019-06-26:
@eslambakr The whole network architecture was redesigned in 0.9.0 to allow having multiple clients and controlling every actor, that wasn't possible the way 0.8.x was implemented.

Try doing `client.get_server_version()` to see if the server is reachable, if that fails then probably is a networking error (firewall, wrong port or used by another app, ...). If you're using synchronous mode is easy to end up in a dead-lock with the server, then eventually the timeout will be met, checkout other issues related to this.

Omaymaq, 2019-07-23:
Hi, I have the same issue at the moment how did you guys sort it out? thank you

JMorceaux, 2019-07-23:
@Omaymaq , I believe IP 0.0.0.0 is a non-valid IP, so just change this by something more conventionnal like 127.0.0.1, your localhost IP.

Omaymaq, 2019-07-23:
@JMorceaux yeah I put that instead of my IP :D

karthi0804, 2019-07-31:
> Is there a way to fix the issue with the server timeouts? The server client connection is lost at random which causes problems in running any longer runs.

@agarwaltanmay  did u find a fix for this?

Omaymaq, 2019-07-31:
@karthi0804 mine worked after I switched off the windows firewall and anti-virus! give it a try if you like :)

sazkooo, 2019-09-15:
You just need to increase 
try:
    client = carla.Client("localhost", 2000)
    client.set_timeout(5.0)

timeout to 5 or something and it will work

karthi0804, 2019-09-24:
> @karthi0804 mine worked after I switched off the windows firewall and anti-virus! give it a try if you like :)
@omaymaq which carla version are u using?

Omaymaq, 2019-09-24:
@karthi0804 0.9.5 copy 

Do you have any idea how can I add my own car to the simulator? thanks :)

KasraMokhtari2112, 2020-04-06:
Hello everyone!

I still have the same issue. Got this error over and over again:
"time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000". 
I tried the mentioned approaches on 2 different python versions. However, they did not work for me. 
I really appreciate it if you can help me with that. Thank you so much!

VEDANTGHODKE, 2020-09-24:
Hello,

I and my team have been trying to implement the GNSS, IMU, RGB Camera and Lidar sensors in Carla. However, since the past 4 days, we have been facing a timeout and connection error on the Carla Unreal Engine.

We have not made any configurational changes to Python or Carla. Can anyone kindly help?

TeeManiac, 2020-11-12:
I am facing this issue. No modifications were made to Carla but getting this error now:
C:\Users\simple\Documents\scenario Runner\scenario_runner>python scenario_runner.py --openscenario srunner\examples\LaneChangeSimple.xosc                                                                                                       Traceback (most recent call last):                                                                                        File "C:\Users\simple\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 598, in <module>                sys.exit(main())                                                                                                      File "C:\Users\simple\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 587, in main                    scenario_runner = ScenarioRunner(arguments)                                                                           File "C:\Users\simple\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 91, in __init__                 self.traffic_manager = self.client.get_trafficmanager(int(self._args.trafficManagerPort))                           RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000 

How can I solve this issue?

TeeManiac, 2020-11-12:
I am facing this issue. No modifications were made to Carla but getting this error now:
C:\Users\dvk\Documents\scenario Runner\scenario_runner>python scenario_runner.py --openscenario srunner\examples\LaneChangeSimple.xosc
Traceback (most recent call last):
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 598, in sys.exit(main())
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 587, in main scenario_runner = ScenarioRunner(arguments)
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 91, in init self.traffic_manager = self.client.get_trafficmanager(int(self._args.trafficManagerPort))
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

How can I solve this issue?

TeeManiac, 2020-11-12:
This issue was resolved after rebooting the system and waiting for some time.

qhaas, 2021-10-01:
There seems to be more than one type of 'timeout' and not all of them respect the value set with the client's `set_timeout` API call.

PR #1741 (2019-06-13): Update/remove unused server timeout
There is no place to set timeout for server in [rpclib](https://github.com/rpclib/rpclib/blob/master/lib/rpc/server.cc)

Closes #1501.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1741)