carla-simulator/carla #1502: Global Time Dilation in Carla

YechengLyu, 2019-04-06:
Hello,

I am wondering if Carla simulator has an attribute to slow down the simulator clock.

Currently, I am working on a project which uses Carla to simulate the world and perceptions and ROS to execute the planning and controller. Unfortunately, my computer is not fast enough to run all modules at real-time speed. If I could slow down the simulator clock similar to the **Global Time Dilation** in **Unreal Engine** or the **-r option** in **rosbag play**, I can check the feasibility of my planning algorithms.

Thanks,
Yecheng

nsubiron, 2019-04-08:
Hi @YechengLyu,

You can run the simulator in synchronous mode, that way the simulator waits until a client calls `world.tick()`. Checkout the docs on [Synchronous mode](https://carla.readthedocs.io/en/latest/configuring_the_simulation/#synchronous-mode), and [synchronous_mode.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/synchronous_mode.py) for an example.

YechengLyu, 2019-04-10:
Thanks. I have figured out the synchronous mode.

closed 2019-04-10 by YechengLyu