carla-simulator/carla #1506: How can I put my camera is a fixed point?[0.9.3]

ChrisHRZ, 2019-04-08:
I need to put one or more camera(depth, RGB and segmentation) in a fixed point to save pictures. I only know how to change the relative position of vehicle. I need to fix my camera or move camera's viewpoint just like in server.How can I achieve this?
Thank you.

nsubiron, 2019-04-15:
Hi @ChrisHRZ,

Just remove the `attach_to` argument when spawning a camera
```py
camera = world.spawn_actor(blueprint, transform)
```

ChrisHRZ, 2019-04-26:
> Hi @ChrisHRZ,
> 
> Just remove the `attach_to` argument when spawning a camera
> 
> ```python
> camera = world.spawn_actor(blueprint, transform)
> ```

Thank you

geekhut, 2019-06-06:
> > Hi @ChrisHRZ,
> > Just remove the `attach_to` argument when spawning a camera
> > ```python
> > camera = world.spawn_actor(blueprint, transform)
> > ```
> 
> Thank you

Hi did you manage to do it? could you share the code please?

closed 2019-08-12 by stale[bot]