carla-simulator/carla #1508: Details regarding spectator

Narendarselva, 2019-04-08:
While going through the docs i came across the term spectator.
1 . What actually it is used for .
2. And also it would be more useful like what are the attributes for each actor that can be changed and explanation of each attributes and usage included into the documents.
for eg : If i create a vehicle actor what are and all parameters (color,dimensions,noofheels)
that are supported in PythonAPI to be changed.
3.And also PythonAPI's with explanation regarding the usage of each would be more helpful.

rgourova, 2019-04-10:
1. From what I have seen, the spectator is used to observe a scene in the server and move around the view there. There is a sample code in Carla 0.9.4 where they use it: vehicle_gallery.py. What they do there is basically showcase the vehicles. I tried moving the spectator along with the car but the car was not moving very smoothly when changing the location of the spectator at each iteration.

Narendarselva, 2019-04-11:
Thanks a lot for useful info

closed 2019-04-11 by Narendarselva