carla-simulator/carla #1517: Unreal crashes when trying to connect with Python API

Abanoub-G, 2019-04-09:
Hi,
I have installed CARLA successfully using the "How to build on Linux " guide provided by you guys. Everything installed just fine.  Apparently this method  does not provide the CarlaUE4.sh file to execute, which is fine because I will need to modify carla for my project.

I launch Carla either through "make launch" or  double clicking the CarlaUE4.uproject project and run play. Then I open another terminal and try to run one of the examples inside the PythonAPI folder, and I end up getting an error ModuleNotFoundError: No module named 'carla'.
So, where can I find the missing module?

Another thing I tried out was to go inside the "Deprecated" folder and then inside the "PythonClient" folder and run the "client_example.py" code. Unreal crashes as soon as I run the  "client_example.py", which I am not sure why it happens.   

Would really appreciate a prompt response.

Thanks

wattsavolt, 2019-04-09:
You need to run "make PythonAPI"

Do the following:
make clean
make launch 
pip install pygame (scripts need pygame)
make PythonAPI (builds the necessary modules)
Go in the unreal editor and in Editor Preferences, uncheck "Use Less CPU when in Background" option hit play in the editor.
Run your script

Now you're good to go.

Abanoub-G, 2019-04-10:
Your are AWESOME!! THANKS A LOT!!! 
That made it work.

closed 2019-04-10 by Abanoub-G