carla-simulator/carla #1518: 0.9.5 no windows pre-compiled version?

TDHTTTT, 2019-04-09:
I couldn't find a pre-compiled 0.9.5 windows version. Where is it? Or is windows support canceled? Thanks!

supermomie, 2019-04-12:
I think the 0.9.5 is not ready yet for Windows [https://github.com/carla-simulator/carla/releases](https://github.com/carla-simulator/carla/releases)

supermomie, 2019-04-12:
You have the 0.9.4 :)

TDHTTTT, 2019-04-12:
> You have the 0.9.4 :)

Thanks! And actually I was using it for my windows machine but the PythonAPI client doe not work (with ddl error)

BTW, I actually saw this issue was mentioned in the changelog for 9.0.5:
> Fixed python client DLL error on Windows

So it is not ready yet?

gunpuz, 2019-04-21:
I am waiting for Windows client DLL error fixes too. 👍

TDHTTTT, 2019-05-04:
It seems it is out: [http://carla-assets-internal.s3.amazonaws.com/Releases/Windows/CARLA_0.9.5.zip](url)

closed 2019-05-04 by TDHTTTT