carla-simulator/carla #1520: Issue running Carla on UE4 with Nvidia Titan V

pparmesh, 2019-04-10:
I was working with Nvidia 1080Ti and the simulator ran at close to 30 FPS. I recently upgraded my GPU to Nvidia Titan V and the simulator's performance has dropped drastically ( close to 3 FPS). Any ideas as to why the drop in performance? My best guess is it is because of the difference in architecture of the two GPUs. If this is the issue, how do I solve it? Thanks!

analog-cbarber, 2019-04-10:
The Titan V is not really intended for graphical rendering, so I would necessarily think you would get any benefit vs a 1089Ti for CARLA, but that is a pretty big difference. Did you recompile UE?

pparmesh, 2019-04-11:
I reinstalled Unreal Engine and Carla but it still is very slow. The built maps are running fine (without Unreal Engine) though.

pparmesh, 2019-04-11:
Issue fixed! Changes made were:
Unreal Engine -> Edit->Editor Preferences->General->Performance-> Unchecked Use less CPU when in background.

closed 2019-04-11 by pparmesh