carla-simulator/carla #1521: Logitech g29 steering issue

aravindSwamy94, 2019-04-10:
Hi , 
         Im using logitech g29 steering wheel and using maual_control_steeringwheel.py from carla release. The problem i face is , steering is calibrated to range of 540 and i see this issue shown in the image. The steering value is extremely high and it goes past the mximum range steering in both directions.

please help me on this regard
[image]
[image]

Johnnyswong, 2019-04-10:
I have implemented an class to control the g29 controller via the joystick module in pygame. The steering wheel is an axis and the gas pedal is a button. 
The thing is, for the steering wheel, the axis perfectly fits the range in the simulator;  for the gas pedal, the value for pressing it is -1 while releasing it is 1, so you need to change the range into 0 to 1. If the value on the button is x, the throttle is:
`-0.5*x+0.5`
A little problem you may also meet is the initial value for the gas pedal seems to 0, so you need to write  some code to catch this case.

You can find everything you need in manual_control.py and https://www.pygame.org/docs/ref/joystick.html

0102TH, 2019-04-15:
，Hello，how about the problem?Has the problem been solvedI also encounter the problem,and haven't dealt it yet. and you ?

aravindSwamy94, 2019-04-15:
Hi , 
          Sorry i didnt get the question completely...But i understood that , u r asking me whether the problem is solved ??
           Actually the problem isnt solved. Im still facing the same issue.

0102TH, 2019-04-15:
I am using Logitech G920 ,and have the same problem with U, but  it seems that the range of wheel shown in the image has no effect on steering wheel control.
[image]

And I have some other problems,the FPS of my client is very low.Do U know the reason ?My cpmperter's GPU is gtx1060,I don't think it's a problem with my computer configuration, but the reason for the program script or map.

0102TH, 2019-04-15:
The fps of this picture is much better than before,because the server fps has become normal.However,client fps is always so low.And I am trying to search ways to solve the problem.Have you encountered a similar problem?IListen with respectful attention.

Xiaoyu006, 2019-05-16:
May I know how you integrate logitech g29 into CARLA simulator on Linux (ubuntu)? Is there a tutorial?

Johnnyswong, 2019-05-16:
Google for the Logitech G29 driver. Than use pygame to setup your own controller. Code can be found in the tutorial code on the pygame website. The KeyboardController class in the manual_control.py also helps.发自我的iPhone------------------ Original ------------------From: Xiaoyu006  <notifications@github.com>Date: Thu,May 16,2019 10:23 AMTo: carla-simulator/carla <carla@noreply.github.com>Cc: JunningHuang <jn8227805@gmail.com>, Comment <comment@noreply.github.com>Subject: Re: [carla-simulator/carla] Logitech g29 steering issue (#1521)May I know how you integrate logitech g29 into CARLA simulator on Linux (ubuntu)? Is there a tutorial?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/carla-simulator/carla/issues/1521?email_source=notifications\u0026email_token=AD4QA43R5O6YHWNM6PKR33LPVTATRA5CNFSM4HEYSWIKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVQO3ZI#issuecomment-492891621",
"url": "https://github.com/carla-simulator/carla/issues/1521?email_source=notifications\u0026email_token=AD4QA43R5O6YHWNM6PKR33LPVTATRA5CNFSM4HEYSWIKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVQO3ZI#issuecomment-492891621",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Johnnyswong, 2019-05-16:
The appendix is my code to connect the Logitech G29. Help it would help.

geekhut, 2019-05-17:
Hi 

> […](#)

I need the code as well, may I know where did you upload it to? thank you!

Xiaoyu006, 2019-05-17:
@Johnnyswong Hi, Johnny! Thanks very much for your reply. I am using the G29 wheel on CARLA0.9.5. I got the example code manual_control_steeringwheel.py run correctly after fixing 2 bugs. 
I searched logitech g29 driver and found there is no official version for use. 
With my experience, I would say that there is no need to have a driver for logitech g29 to be used in linux.

Xiaoyu006, 2019-05-17:
> Hi
> 
> > […](#)
> 
> I need the code as well, may I know where did you upload it to? thank you!

Which version of CARLA are you using?

Xiaoyu006, 2019-05-17:
> Hi ,
> Im using logitech g29 steering wheel and using maual_control_steeringwheel.py from carla release. The problem i face is , steering is calibrated to range of 540 and i see this issue shown in [...]

try to change this line in your maual_control_steeringwheel.py file
        steerCmd = K1 * math.tan(1.1 * jsInputs[self._steer_idx])
to:
        steerCmd = K1 * math.tan(0.8 * jsInputs[self._steer_idx])

It works for me and you can adjust the value by yourself.

geekhut, 2019-05-17:
Hi all, I got this error after running manual_control_steeringwheel.py vanilla:

_configparser.NoSectionError: No section: 'G29 Racing Wheel'_

I have connected the wheel to my ubuntu PC, but without any drivers. Will it work? Thanks.

Xiaoyu006, 2019-05-17:
> Hi all, I got this error after running manual_control_steeringwheel.py vanilla on a pre-compiled carla:
> 
> _configparser.NoSectionError: No section: 'G29 Racing Wheel'_
> 
> I have connected the wheel to my ubuntu PC, but without any drivers. Will it work? Thanks.

You first find where the line:
        self._parser.read('wheel_config.ini')
then you find the file 'wheel_config.ini', which should be in the carla folder.
        replace the file name in the code by its absolute path, will solve this bug.

Derekabc, 2019-05-17:
@Xiaoyu006 it seems i cannot find the 'wheel_config.ini' in the carla folder. Where do you find the file? or we need to create a new one. Thanks

Derekabc, 2019-05-17:
hi, what should i modify to use the G920 using manual_control_steeringwheel.py? Thanks

Xiaoyu006, 2019-05-18:
Hi, @Derekabc. I got this code run properly with G29 : https://github.com/Xiaoyu006/carla/blob/master/PythonAPI/examples/test_g29.py
You may run this code with your g920 first. Maybe the *config.ini file should be changed since the difference between g29 and g920

morningsky, 2019-05-20:
@Xiaoyu006  What's the difference between your code and  manual_control_steeringwheel.py ?
I can use G29 control the car, but it has no force feedback effect, have you met the same problem?
My environment: Ubuntu18.04, Carla 0.9.5.

Xiaoyu006, 2019-05-21:
@morningsky No big diff. just fixed small bugs like this 
            # 'Map:     % 20s' % world.world.map_name,
            # Xiaoyu changed here because the update from carla0.9.3 to carla0.9.4
            'Map:     % 20s' % world.world.get_map().name,.

We are running CARLAJ 0.9.5 on Ubuntu 16.04 with logitech G29 and do not have force feedback effect in PS3 mode. 
Sometimes replug-in the wheel usb will give the force but we have not figured out why.
Do you have any ideas?

geekhut, 2019-05-21:
> You first find where the line:
> self._parser.read('wheel_config.ini')
> then you find the file 'wheel_config.ini', which should be in the carla folder.
> replace the file name in the code by its absolute [...]

Thanks that helped!

morningsky, 2019-05-21:
> @morningsky No big diff. just fixed small bugs like this
> # 'Map: % 20s' % world.world.map_name,
> # Xiaoyu changed here because the update from carla0.9.3 to carla0.9.4
> 'Map: % 20s' % [...]

 No.  My G29  does not have force feedback effect  in PS3 mode, and I still can not solve this problem...

geekhut, 2019-05-21:
> No. My G29 does not have force feedback effect in PS3 mode, and I still can not solve this problem...

Just reconnect your G29 to your PC a few times, the force feedback should be initialised upon connection.

morningsky, 2019-05-21:
@geekhut  You means reinstall jstest-gtk? Have you succeeded in this way? My system is Ubuntu18.04, Is it ok?  Thanks!!!

geekhut, 2019-05-21:
> @geekhut You means reinstall jstest-gtk? Have you succeeded in this way? My system is Ubuntu18.04, Is it ok? Thanks!!!

Nope, it has nothing to do with jstest-gtk, just plug and un-plug your G29 a few times until the force feedback works.

morningsky, 2019-05-22:
@geekhut  I tried many times, but it's doesn't work

morningsky, 2019-05-22:
@geekhut  It runs a few seconds when I plug my G29 USB interface, then the force feedback effect can't work.

morningsky, 2019-05-22:
> @morningsky No big diff. just fixed small bugs like this
> # 'Map: % 20s' % world.world.map_name,
> # Xiaoyu changed here because the update from carla0.9.3 to carla0.9.4
> 'Map: % 20s' % [...]

Have you solved this problem?

0102TH, 2019-05-22:
@Derekabc Hi，have u pluged G920 into the carla?Any problems have you met?Actually ,G920 is the same as G29,the config.ini need be changed ,this need u test the wheel in person.

morningsky, 2019-05-22:
> @geekhut It runs a few seconds when I plug my G29 USB interface, then the force feedback effect can't work.

@geekhut  I solved it by tried more times....... Thanks!

0102TH, 2019-06-01:
@Xiaoyu006 
Thank u for your sharing .I made it with changing that line in maual_control_steeringwheel.py.
Now,the range of steer in pygame is nomal .

closed 2019-08-07 by stale[bot]

queen-king, 2020-02-20:
Hi all. 
I am running carla0.9.5 in windows. Will i be able to integrate g29 with windows.

ujjawal-kumar, 2020-03-09:
> Hi all.
> I am running carla0.9.5 in windows. Will i be able to integrate g29 with windows.

Yes, I am successfully run with carla0.9.5 in windows 10. only you have to change in
manual_control_steeringwheel.py as described by @Xiaoyu006  
replace "world.world.map_name" to "world.world.get_map().name"
 in tick function in line number 427.

seuwcs, 2021-04-04:
Dear all friends,
I am connecting the logitech g29 to carla (version 0.9.10) on Linux Ubuntu 18.04, but something went wrong after the connection was made. My confusion is that: the car generated in the scene by the script "manual_control_steeringwheel.py " could not be controlled by the steering wheel as well as the pedal.

The vehicle remains still not matter which pedal I stepped, the instruction said that I should firstly press the brake pedal, but it doesn't work, the value of the brake pedal remain unchanged . 
[image]
Who can give me some suggestions? any advice will be appreciated greatly.

ujjawal-kumar, 2021-04-04:
In Logitech g29 you have to press handbrake.

seuwcs, 2021-04-04:
> In Logitech g29 you have to press handbrake.
> […](#)

Thanks very much for your addvice, but the Logitech g29 does not have a handbrake steer.

And I have tried to change the value of handbrake in the file of wheel_config.ini

[G29 Racing Wheel]
steering_wheel = 0
throttle = 2
brake = 3
reverse = 5
handbrake = 4

Changed it to 0, but it could not help, which confused me a lot.

Or should I buy another Logitech g29 steering wheel  and have a try ?

ujjawal-kumar, 2021-04-04:
Hi,
Can you First separately checked your Logitech g29 steering wheel with
pygame event. Which input button is giving what input index event when you
pressed that button. then you can easily mapped those button.

seuwcs, 2021-04-04:
Hi, thanks for your advice. I'll have a try later.  very much thanks. 

 
 
 &gt; In Logitech g29 you have to press handbrake.
 &gt; … <#m_-462894048820574199_&gt;
 &gt; On Sun, 4 Apr 2021, 17:15 Changshuai Wang, *@*.***&gt; wrote: Dear all
 &gt; friends, I am connecting the logitech g29 to carla (version 0.9.10) on
 &gt; Linux Ubuntu 18.04, but something went wrong after the connection was made.
 &gt; My confusion is that: the car generated in the scene by the script
 &gt; "manual_control_steeringwheel.py " could not be controlled by the steering
 &gt; wheel as well as the pedal. The vehicle remains still not matter which
 &gt; pedal I stepped, the instruction said that I should firstly press the brake
 &gt; pedal, but it doesn't work, the value of the brake pedal remain unchanged .
 &gt; [image: 微信图片_20210403175649]
 &gt; https://user-images.githubusercontent.com/65322137/113507613-2ee35380-957e-11eb-899d-31be273acd7a.png
 &gt; Who can give me some suggestions? any advice will be appreciated greatly. —
 &gt; view it on GitHub <#1521 (comment)
 &gt; <https://github.com/carla-simulator/carla/issues/1521#issuecomment-813019183&gt;&gt;,
 &gt; .
 &gt;
 &gt; Thanks very much for your addvice, but the Logitech g29 does not have a
 &gt; handbrake steer.
 &gt;
 &gt; And I have tried to change the value of handbrake in the file of
 &gt; wheel_config.ini
 &gt;
 &gt; [G29 Racing Wheel]
 &gt; steering_wheel = 0
 &gt; throttle = 2
 &gt; brake = 3
 &gt; reverse = 5
 &gt; handbrake = 4
 &gt;
 &gt; Changed it to 0, but it could not help, which confused me a lot.
 &gt;
 &gt; Or should I buy another Logitech g29 steering wheel and have a try ?
 &gt;
 &gt; —
 &gt; <https://github.com/carla-simulator/carla/issues/1521#issuecomment-813051029&gt;,
 &gt; .
 &gt;

alibukharai, 2021-04-09:
> Hi, @Derekabc. I got this code run properly with G29 : https://github.com/Xiaoyu006/carla/blob/master/PythonAPI/examples/test_g29.py
> You may run this code with your g920 first. Maybe the *config.ini file should be changed since the difference between g29 and g920

Hi, I am new to carla, I am trying to manually control the vehicle in carla with g29 steering wheel in a windows environment. But go the same error, "No section G29 Racing wheel" all the time when tried. Can you help me with this

yeyu2020, 2021-09-24:
Hi , I get the same issue with you , now what's going on. If you have solved it, maybe you can give me some adivce . Thank you very much.

kaka-lin, 2022-05-27:
> Dear all friends, I am connecting the logitech g29 to carla (version 0.9.10) on Linux Ubuntu 18.04, but something went wrong after the connection was made. My confusion is that: the car generated in [...]

Hi, I got the same issue, have any update?

FBanani, 2022-06-29:
> 

Hi
have you fixed it? because I recently want to work with Carla I bought wheels and pedals G29 and I want to run it in windows 10? I have install Logitech G29 hub. what should I do for the next step?

FBanani, 2022-06-29:
> Hi, I am new to carla, I am trying to manually control the vehicle in carla with g29 steering wheel in a windows environment. But go the same error, "No section G29 Racing wheel" all the time when [...]

FBanani, 2022-06-29:
> Hi
> have you fixed it? because I recently want to work with Carla I bought wheels and pedals G29 and I want to run it in windows 10? I have install Logitech G29 hub. what should I do for the next step?

FBanani, 2022-06-29:
#

> Yes, I am successfully run with carla0.9.5 in windows 10. only you have to change in manual_control_steeringwheel.py as described by @Xiaoyu006 replace "world.world.map_name" to [...]

Hi
have you fixed it? because I recently want to work with Carla I bought wheels and pedals G29 and I want to run it in windows 10? I have install Logitech G29 hub. what should I do for the next step?