carla-simulator/carla #1522: Adding noise to the autopilot control in CARLA 0.9.X

marintoro, 2019-04-10:
I would like to perturb the autopilot control with some noise (both for steering and for throttle).
How would that be possible? Is that possible to perturb with different noise for each actor with autopilot set to True?

I know it was possible in CARLA 0.8.X (see the issue #780 ) and I couldn't find how to make it with the new API.

xmyqsh, 2019-06-09:
@marintoro 
You can do it in the world on_tick() callback:
```
control = vehicle.get_control()
control.throttle += random.uniform(-0.1, 0.1)
control.steer += random.uniform(-0.1, 0.1)
vehicle.apply_control()
```
Note that: [VehicleInputPriority.h](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h#L13-L24)
```
 13 UENUM(BlueprintType)
 14 enum class EVehicleInputPriority : uint8
 15 {
 16   INVALID = 0u  UMETA(Hidden),
 17
 18   Lowest        UMETA(DisplayName = "Lowest Priority", ToolTip = "Use for debugging purposes only"),
 19   Relaxation    UMETA(DisplayName = "Control Relaxation Input", ToolTip = "Control relaxation when no other input is provided (non-sticky control)"),
 20   Autopilot     UMETA(DisplayName = "Autopilot Input", ToolTip = "Input provided by the built-in autopilot"),
 21   User          UMETA(DisplayName = "User Input", ToolTip = "Input provided by an user playing in the simulator"),
 22   Client        UMETA(DisplayName = "Client Input", ToolTip = "Input provided by an RPC client connected to the simulator"),
 23   Highest       UMETA(DisplayName = "Highest Priority", ToolTip = "Use for debugging purposes only")
 24 };
```

The Priority of Client is higher than Autopilot.

eslambakr, 2019-06-12:
@xmyqsh 
Hello,
I am facing this issue too, and unfortunately your solution didn't work with me.

Did you try it? (I am using CARLA 9.5 if it is matter)

Thanks in advance.

closed 2019-08-18 by stale[bot]