carla-simulator/carla #1524: Trouble building CARLA on Ubuntu  18.04

Canorvantis, 2019-04-10:
I am currently trying to set up Carla on a freshly installed Ubuntu 18.04, following the instructions "How to build on Linux". I have encountered two issues (I don't know yet if they are related in any way): 

- libpng16-dev can not be installed. German error message: 

libpng16-dev : Hängt ab von: libpng16-16 (= 1.6.20-2) aber 1.6.34-1ubuntu0.18.04.1 soll installiert werden
                Empfiehlt: libpng16-tools soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.

- When I hit "make launch", UE4 always freezes (mostly at exactly 94%),  exactly ONCE the Editor actually opened (but crashed soon after), sometimes it occurs at an earlier loading stage. 

Has anyone encountered this issue and found a solution?

 Thank you

Johnnyswong, 2019-04-10:
Try the compiled version ?

Iswed, 2019-04-11:
Hi, I've just built Carla on Ubuntu 18.04 and it works just fine. I was given the warning that libpng16-dev has been replaced by libpng-dev. So I installed that instead.
Also make sure you're using the UE4.21, as UE4.22 has been released and it currently won't work with CARLA.
Also make sure you compile both UE4Editor and CARLA with clang 6 compiler.

supermomie, 2019-04-12:
Hello @Canorvantis 
What version of clang you have ?
Have you followed these instructions ? [here](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/)

supermomie, 2019-04-12:
For ubuntu 18.04.* you need to install libpng16-16, not libpng16-dev.
`sudo apt install libpng-16-16`

And for the crash at 94% can you share what's the error(s) message(s) said on your terminal ?

Canorvantis, 2019-04-15:
First of all, thank you guys for all your answers and suggestions.

I strictly followed the instructions on the CARLA page, so I suppose compilers etc. are set right.

I think it could also have to do with my graphics driver, I get a warning from UE4 that the driver (Nouveau for GTX1050) is unknown and that could lead to trouble. 

I never get a warning message in the terminal, the whole system freezes and I am forced to restart it through the power button. But most of the times (if not everytime), the last terminal entries I get have to do with loading models for cars.  I have already tried the NVIDIA drivers, but they caused another trouble with screen resolution etc. so I went back to the default version.

closed 2019-06-21 by stale[bot]