carla-simulator/carla #1526: Why d_angle < 90.0?

TDHTTTT, 2019-04-11:
Hi! As I am trying to figure out how did carla implement the autopilot, I came across this helper function in ```/carla/PythonAPI/carla/agents/tools/misc.py``` :
```
def is_within_distance_ahead(target_location, current_location, orientation, max_distance):
    """
    Check if a target object is within a certain distance in front of a reference object.

    :param target_location: location of the target object
    :param current_location: location of the reference object
    :param orientation: orientation of the reference object
    :param max_distance: maximum allowed distance
    :return: True if target object is within max_distance ahead of the reference object
    """
    target_vector = np.array([target_location.x - current_location.x, target_location.y - current_location.y])
    norm_target = np.linalg.norm(target_vector)

    # If the vector is too short, we can simply stop here
    if norm_target < 0.001:
        return True

    if norm_target > max_distance:
        return False

    forward_vector = np.array(
        [math.cos(math.radians(orientation)), math.sin(math.radians(orientation))])
    d_angle = math.degrees(math.acos(np.dot(forward_vector, target_vector) / norm_target))

    return d_angle < 90.0
```
For the most part, it is straightforward, if the vector between me and target is short it returns True and False if it is larger than a threshold.

However near the end it calculates the angle between the forward_vector and the target_vector and returns if d_angle < 90.0. Why doing that? Does that mean if the previous vector is not too short or too long, we try to see if we are facing that vehicle? If we are facing it, than we still consider is_within_distance_ahead == True?

bernatx, 2019-04-15:
Hi,

As I see the function is called 'is_within_distance_ahead', so it seems to only apply for targets ahead from you. If the angle (dot product) is bigger than 90 degrees then the target is behind you, so not valid for this function. Does that make sense?

Regards

TDHTTTT, 2019-04-17:
> Hi,
> As I see the function is called 'is_within_distance_ahead', so it seems to only apply for targets ahead from you. If the angle (dot product) is bigger than 90 degrees then the target is behind you, so not valid for this function. Does that make sense?
> Regards

Yes but why comparing the norm_target to 0,01 in the first place? Isn't it a bit redundant?

closed 2019-04-19 by TDHTTTT