carla-simulator/carla #1529: PythonAPI carla.Waypoint.s?

TDHTTTT, 2019-04-12:
So there is not enough documentation in the PythonAPI and I couldn't figure out what Carla.Waypoint.s is.

TDHTTTT, 2019-04-12:
It seems that it is the definition from open drive, which is "track position of object's origin (s-position)", which still does not make sense to me. What is the origin?

TDHTTTT, 2019-04-12:
I also found this line in the update for 0.9.5 

> "The s parameter of OpenDrive, it defines the position along the reference line, measured in meters from the beginning of the track, calculated in the XY-plane"

I guess origin == beginning of the track, but what does it mean?

nsubiron, 2019-04-15:
Take a look at https://github.com/carla-simulator/carla/issues/1469#issuecomment-478330987, may help understanding OpenDrive format.

TDHTTTT, 2019-04-15:
> Take a look at #1469 (comment), may help understanding OpenDrive format.

Thanks!

closed 2019-04-15 by TDHTTTT