carla-simulator/carla #1532: [0.9.5] when using the command "make launch"   there is an error

sheepwxy, 2019-04-12:
hello. I  want to install Carla on my computer  as the instructions by the DOCS, however, after I use the command "make launch",there is an error as below:

> Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
Setup.sh: googletest-1.8.0-ex already installed.
fatal: Not a git repository (or any of the parent directories): .git
Util/BuildTools/Linux.mk:101: recipe for target 'setup' failed
make: *** [setup] Error 128

my Ubuntu Version is 16.04  Unreal Version is 4.21.

ps: I have doubted that the error is caused by the older version in my computer. So i remove the older Carla0.9.1 and the older Unreal 4.19.  However it does not work.

supermomie, 2019-04-12:
Hello @sheepwxy have you try after removing the old versions (Carla && Unreal) to rebuild ?
`make rebuild`

sheepwxy, 2019-04-12:
thanks for your replying, but there is an new error

> fatal: bad revision 'HEAD'
Util/BuildTools/Linux.mk:101: recipe for target 'setup' failed
make: *** [setup] Error 128

supermomie, 2019-04-12:
You lost you're `.git` folder I think.
Did you clone the repo or copy paste from somewhere ? You're first error message said you've no folder `.git`, and for your second message no "HEAD" (git).

Try to reclone the repo -> `./Update.sh` -> `export UE4_ROOT=~/UnrealEngine_4.21` -> `make launch`

sheepwxy, 2019-04-13:
> You lost your `.git` folder I think.
> Did you clone the repo or copy paste from somewhere ? You're first error message said you've no folder `.git`, and for your second message no "HEAD" (git).
> 
> Try to reclone the repo -> `./Update.sh` -> `export UE4_ROOT=~/UnrealEngine_4.21` -> `make launch`
thanks very much! your reply solved my problem!

closed 2019-04-13 by sheepwxy

SBM1952, 2020-02-24:
Hi,
I've followed the steps above an i m getting these prompts:

After Update.sh

Content.tar.gz      100%[===================>]   9,25G  5,93MB/s    in 27m 19s 

2020-02-24 09:00:44 (5,78 MB/s) - ‘Content.tar.gz’ saved [9936321834/9936321834]
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

after building the target:
(base) sbm_linux@sbmlinux:~/carla$ export UE4_ROOT=~/UnrealEngine_4.21 
(base) sbm_linux@sbmlinux:~/carla$ make relaunch
make: *** No rule to make target 'relaunch'.  Stop.
(base) sbm_linux@sbmlinux:~/carla$ make launch
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
cp: cannot create regular file 'boost-1.69.0-c7-install/include/boost/rational.hpp': No such file or directory
Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 1

Other inputs:
OS: ubuntu 18.04.04 LTS
Some inputs: cmake version 3.15.2
Carla branch: master
I tried switching to stable branch and rebuild but no effect

ScaramuzzinoGiovanna, 2020-02-25:
Hi, 
after running 'make launch', using:

_OS: ubuntu 18.04.04 LTS
cmake: version 3.10.2
Carla branch: master
Unreal_engine: 4.22_

The following error is shown:

_Setup.sh: Extracting boost for Python 2.
Building Boost.Build engine with toolset clang... tools/build/src/engine/b2
Unicode/ICU support for Boost.Regex?... not found.
Generating Boost.Build configuration in project-config.jam for clang...

sh: 1: clang++: not found
carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang-linux.jam:63: in clang-linux.init
*** argument error
* rule clang.init-cxxstd-flags ( toolset : condition * : version )
* called with: ( clang-linux : <toolset>clang-linux :  )
* missing argument version
carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang.jam:54:see definition of rule 'clang.init-cxxstd-flags' being called
carla/Build/boost-1.72.0-c7-source/tools/build/src/build/toolset.jam:44: in toolset.using
carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang.jam:33: in clang.init
carla/Build/boost-1.72.0-c7-source/tools/build/src/build/toolset.jam:44: in toolset.using
carla/Build/boost-1.72.0-c7-source/tools/build/src/build/project.jam:1104: in using
project-config.jam:12: in modules.load
carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:255: in load-config
carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:486: in load-configuration-files
carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:607: in load
carla/Build/boost-1.72.0-c7-source/tools/build/src/kernel/modules.jam:295: in import
carla/Build/boost-1.72.0-c7-source/tools/build/src/kernel/bootstrap.jam:139: in boost-build
carla/Build/boost-1.72.0-c7-source/boost-build.jam:17: in module scope
Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 1_