carla-simulator/carla #1533: PyGame bad performance

klemen666, 2019-04-12:
I open Carla server on one computer and i open python client on another (py manual_control.py) which opens PyGame. In the HUD it says both server and client are running with 60fps, but the rendering in client is bad - seems like 15 fps. What could be the reason? Network between computers is 1Gbit. I'm using Carla 0.9.4 on Windows.

0102TH, 2019-04-19:
@klemen666 
Maybe I have the same problem with U.And I haven't dealt it yet.
Do U run the map of carla own  or other maps?There is one tip is that when the carla server is runnning,never let the server in the background.That's to say,when the server running,the mouse and keyboard can only move the map's perspective and location of camera,cannot operate other programs.Otherwise,the fps is low.

Xiaoyu006, 2019-05-17:
> I open Carla server on one computer and i open python client on another (py manual_control.py) which opens PyGame. In the HUD it says both server and client are running with 60fps, but the rendering [...]

when I ran manual_control.py using carla 0.9.5, 

the fps of server is about 15 fps.
that of client is about 60fps, which is good enough.

But when i ran manual_control_steeringwheel.py on carla 0.9.5. 
the fps of server is about 15 fps.
that of client is about 10fps, which is bad enough.

Not fixed now.

0102TH, 2019-06-01:
Now I see.In carla,we can set it to run in the background without lowering the cpu.Open 'Edit'-->'edit performance'-->'performance'-->'use less cpu when in background'.

closed 2019-08-07 by stale[bot]