carla-simulator/carla #1536: Low FPS when I run manual_control_sreeringwheel.py in ubuntu 16.04,and the driving wheel is Logitech G920

0102TH, 2019-04-15:
system:Ubuntu 16.04
gpu:GTX1060
version:CARLA 0.9.5

Description:
One map of carla is running at other conputer,the Logitech G920 driving force wheel is plugged into my computer ,when I run manual_control_steeringwheel.py,the FPS of Sever and Client is so low that the car's running is not smooth enough.FPS of them is respectively about 16,16.
I am full of confusion.Can anyone give some suggestion?Thanks in advance

sangeetsu, 2019-04-16:
I am facing the same issue with an RTX 2080Ti, Xeon and 32Gigs of RAM. I am getting about 20FPS on the steering wheel adaptation compared to 60FPS on manual_control.py
Any help with this would be greatly appreciated. Thank you!

0102TH, 2019-04-17:
> I am facing the same issue with an RTX 2080Ti, Xeon and 32Gigs of RAM. I am getting about 20FPS on the steering wheel adaptation compared to 60FPS on manual_control.py
> Any help with this would be greatly appreciated. Thank you!
Does your map running on the server come with it?The map I am running is what we draw.However,when I run the map of carla own on other computer,I run manual_control.py or manual_control_steeringwheel.py,both fps are better than running our own map.

Do you  run the map（that is server) and client on the same computer?When running the server and client on the same computer, first run the map and then run the client. However, when running the client, the server is now in the background by default, and the resource response allocated by the computer to the server is reduced.

0102TH, 2019-04-17:
@sangeetsu 
Does your map running on the server come with it?The map I am running is what we draw.However,when I run the map of carla own on other computer,I run manual_control.py or manual_control_steeringwheel.py,both fps are better than running our own map.

Do you run the map（that is server) and client on the same computer?When running the server and client on the same computer, first run the map and then run the client. However, when running the client, the server is now in the background by default, and the resource response allocated by the computer to the server is reduced.

sangeetsu, 2019-04-17:
I've tried with both the packaged map in the precompiled version, as well as maps that I made changes to. The thing is I'm getting good enough frame rates (60-70FPS) on manual_control.py. But the frame rates are abysmal (6-25) for manual_control_steeringwheel.py. And I'm running both server and client on the same pc. I'm not sure how to connect the client to a server running on another pc on the same network. I tried this:
`python manual_control.py --host=<your-ip-address-here>`
But it doesn't seem to detect the server.

0102TH, 2019-04-17:
@sangeetsu 
In terminal enter the folder where the program is located,and then enter:
python manual_control.py  --host <ip of the other computer>
So it is: manual_control_steeringwheel.py  --host <ip of the other computer>

0102TH, 2019-04-21:
@sangeetsu 
I'm sorry that I said the wrong way to connect the other pc.
How about the problem?
In Ubuntu,in terminal enter the folder where the program is located,and then enter:
./manual_control.py  --host <ip of server>
or enter:
./manual_control_steeringwheel.py  --host <ip of server>

In windows,in cmd enter the CarlaPythonAPI,and then enter:
python manual_control.py  --host <ip of server>

Xiaoyu006, 2019-05-17:
the same issue with 
system:Ubuntu 16.04
gpu:  GTX1080 * 2
version:  CARLA 0.9.5
cpu: Xeon

0102TH, 2019-06-02:
One problem:
when I replace the " 'Map:     % 20s' % world.map.name," in manual_control.py with "'Map:     % 20s' % world.world.get_map().name," in the manual_control_steeringwheel.py,and then ,run the manual_control.py,the fps of client in the pygame is very low as same as running manual_control_steerinwheel.py.

0102TH, 2019-06-06:
I finally found the reason,
in the folder named opendrive in carla,we copied one file only 1kb size,changed the name to the map's name,and then ,runned the map ,plugined the wheel,the fps of client will be much better.

jd401, 2019-06-12:
> I finally found the reason,
> in the folder named opendrive in carla,we copied one file only 1kb size,changed the name to the map's name,and then ,runned the map ,plugined the wheel,the fps of client will be much better.

can you tell more detail please,copy whitch file?i have the same issue :(

0102TH, 2019-06-13:
@jd401 
In the folder named opendrive in the folder of carla,there are some files ,their type is icon.cCan U find anyone just 1kb size?
If yes,copy one and rename it with the map's name you want to run.
If not,I can give u one .

jd401, 2019-06-20:
> @jd401
> In the folder named opendrive in the folder of carla,there are some files ,their type is icon.cCan U find anyone just 1kb size?
> If yes,copy one and rename it with the map's name you want to run.
> If not,I can give u one .
sorry,i can't find icon type files in opendrive folder,just one .cpp type file and one .h type file :( , can you give me one please? i've being disturbed by this issue for a long time,i appreciate your help, tank you so much, i email address is hkjxzeiwang55@163.com.

sangeetsu, 2019-06-20:
> @jd401
> In the folder named opendrive in the folder of carla,there are some files ,their type is icon.cCan U find anyone just 1kb size?
> If yes,copy one and rename it with the map's name you want to run.
> If not,I can give u one .

Could you explain a little better? I'm facing the same issue, haven't got decent fps for a while now.
If you're sending the file, kindly send me a copy as well - sangeetulhas@gmail.com

jd401, 2019-06-20:
@0102TH
In the folder named opendrive in the folder of carla,there are some files ,their type is icon.cCan U find anyone just 1kb size?
If yes,copy one and rename it with the map's name you want to run.
If not,I can give u one .

sorry,i can't find icon type files in opendrive folder,just one .cpp type file and one .h type file :( , can you give me one please? i've being disturbed by this issue for a long time,i appreciate your help, tank you so much, i email address is hkjxzeiwang55@163.com.

jd401, 2019-06-20:
@0102TH
In the folder named opendrive in the folder of carla,there are some files ,their type is icon.cCan U find anyone just 1kb size?
If yes,copy one and rename it with the map's name you want to run.
If not,I can give u one .

sorry,i can't find icon type files in opendrive folder,just one .hpp type file and one .h type file :( , can you give me one please? i've being disturbed by this issue for a long time,i appreciate your help, thank you so much, my email address is hkjxzeiwang55@163.com.

0102TH, 2019-06-24:
I'm sorry that I've described incorrectly.In the folder named opendrive in the folder of carla,there are some files ,their type is xodr.
[image]
I uploaded it to my account.Just change the name into the map's name of yours.It will be better.

jd401, 2019-06-26:
> I'm sorry that I've described incorrectly.In the folder named opendrive in the folder of carla,there are some files ,their type is xodr.
> [image]
> I uploaded it to my account.Just change the name into the map's name of yours.It will be better.

thank you,it does run better!!but when i run 0.9.5 version town03 with 40 vehicles spawn,fps is still not hgih enough,maybe it's because my gpu is 1050ti.thank you!

0102TH, 2019-06-26:
@jd401 You are welcome.
Maybe it is because of  the gpu.
In the case of adding a small amount of spawn vehicles, fps is still greatly improved.

kbarbora, 2019-08-06:
Where can I get that file? 
@0102TH you dont have any repos in your github

0102TH, 2019-08-06:
@kbarbora I just uploaded one small file (.xodr) in my github.

kbarbora, 2019-08-06:
Great, thanks. Im getting 60fps.

wangchengkang, 2019-08-11:
Hi，I downloaded the file from your github, changed the name into ‘Town03.xodr’,  and replace my initial Town03.xodr in the opendrive folder with the new one. However when I run manual_control_steeringwheel.py. I still only get 10FPS.
Do you have any idea why the FPS didn’t improve.
Thanks!

wangchengkang, 2019-08-12:
> Hi，I downloaded the file from your github, changed the name into ‘Town03.xodr’, and replace my initial Town03.xodr in the opendrive folder with the new one. However when I run manual_control_steeringwheel.py. I still only get 10FPS.
> Do you have any idea why the FPS didn’t improve.
> Thanks!
My version is 0.9.6 with G29
I have found where the problem lies. The modification of ‘world.map.map_name’ into ‘world.world.get_map()’ will consume lots of computing resources. 
If you don’t care so much about the map’s information in your HUD, just comment this line will fix the bug. Or you can learn from the munaul_control.py, in the World class, add variable ‘self.map’, and make adaptions of the remaining code.

0102TH, 2019-08-12:
> My version is 0.9.6 with G29
> I have found where the problem lies. The modification of ‘world.map.map_name’ into ‘world.world.get_map()’ will consume lots of computing resources.
> If you don’t care so [...]

@wangchengkang 
Hi，u are right.Chaging  ‘world.map.map_name’ into ‘world.world.get_map()’ will consume lots of computing resources.But ,if I do not do this,the program will not run successfully.As you said comment the line,I will try it later.

closed 2019-10-18 by stale[bot]

coderinmaking, 2020-05-15:
just like manual_control.py, create self.map attribute in World class itself like this-

```
try:
    self.map = self.world.get_map()
except RuntimeError as error:
    print('RuntimeError: {}'.format(error))
    print('  The server could not send the OpenDRIVE (.xodr) file:')
    print('  Make sure it exists, has the same name of your town, and is correct.')
    sys.exit(1)
```

In the class HUD, inside method tick, 

in self._info_text change Map attribute as follows-

```
'Map:     % 20s' % world.map,
# 'Map:     % 20s' % world.world.get_map().name,
```

This is giving me 60 fps for client and server being in different systems.
Client: Windows OS, 1060 Nvidia GPU, i7
Server: Windows OS, 1080 Nvidia GPU, i7
Frame resolution set : 720x480