carla-simulator/carla #1538: Automatically generating Walker Spawn Points on map in 0.9.5

FrancescoPinto, 2019-04-15:
Hi,
If I am not wrong, in previous versions of CARLA it was possible to automatically spawn walkers in the world. However, in 0.9.5 it seems to be no more possible to do this automatically. I have modified the source of Carla to retrieve walker spawn points, however, the problem is that placing those spawn points manually around the maps is quite time-demanding. Is it possible to write a python script that randomly places spawn points on zebra crossings and sidewalks? Alternatively, is there any other way to do this programmatically?

hongyuli, 2019-04-16:
Same needs here. Hope this could be considered as a new feature.

ThierryDubois, 2019-05-22:
Same here. I am trying to spawn walkers with automatic movement, but right now they are spawning at the vehicule spawn points and the autopilot does not work.

closed 2019-07-28 by stale[bot]