carla-simulator/carla #1542: Import Mapbox map as Carla map

zx627, 2019-04-16:
Hi
I would like ask if it is possible to import 3D Mapbox map into Carla as map?
If it is possible,please explain.

Thank you.

XGodina, 2019-04-16:
Hi,
I don't know wich type of format generates the "3D Mapbox". But, if it is .fbx or .obj, you can export to Carla.

zx627, 2019-04-19:
> Hi,
> I don't know wich type of format generates the "3D Mapbox". But, if it is .fbx or .obj, you can export to Carla.

Thank you for your reply.
Yes.it is .fbx, but is there a reference document about how to export to Carla?

XGodina, 2019-04-20:
Hi,

There is our documentation "https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/". But currently we working to update this doc. If you need somthing you can use githab or discord to ask whatever you want.

zx627, 2019-04-24:
> Hi,
> 
> There is our documentation "https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/". But currently we working to update this doc. If you need somthing you can use githab or discord to ask whatever you want.

Thank you for your help.
I have read the documentation about how to make a new map.
The document says that there are two kinds of files are necessary ，fbx and xodr.
But I have only fbx file generated from Unity but have no xodr file.
Can I use only fbx files export to Carla?

XGodina, 2019-04-24:
The .xodr file is to refer to OpenDrive information.
Yes,  You can use only the .fbx files but keep in mind you only drive in manual_control.py.

Derekabc, 2019-04-26:
Does that mean we can only control car manually if we do not use the OpenDrive actor?

0797liyn, 2019-04-29:
Hi,

When i import .fbx files into carla(follow the Step 2 - Unreal of your of documentation "https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/"), it remind same errors. 

> No smoothing group information was found in this FBX scene.  Please make sure to enable the 'Export Smoothing Groups' option in the FBX Exporter plug-in before exporting the file.  Even for tools that don't support smoothing groups, the FBX Exporter will generate appropriate smoothing data at export-time so that correct vertex normals can be inferred while importing. 
No polygon were found on mesh  'Scene' 
No polygon were found on mesh  'Scene' 
> No polygon were found on mesh  'Scene' 

But when i try to solve it by export .fbx files again, i can not find any options about 'Export Smoothing Groups' with my exporter(i use Unity FBX Exporter)
[image]
 
 If i aim to drive in manual_control.py on my .fbx map, how to import without errors?
Thank you.

0797liyn, 2019-05-06:
@XGodina When i import .fbx files and run manual_control.py, it crash. 
Which lines or parts of manual_control.py should i modify ,to not load .xodr? 
(I only want to work manual)

>[2019.05.06-08.52.36:311][671]LogCarla: FOpenDrive: Corrected map name from NewMap to NewMap
[2019.05.06-08.52.36:311][671]LogCarla: Loading OpenDrive file '../../../../carlaMap/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/NewMap.xodr'
[2019.05.06-08.52.36:311][671]LogTemp: Error: Failed to load OpenDrive file '../../../../carlaMap/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/NewMap.xodr'
[2019.05.06-08.52.36:978][673]LogCarla: Spawning actor 'vehicle.mini.cooperst'
[2019.05.06-08.52.36:979][673]LogBlueprintUserMessages: [Mini_C_0] MaterialNotFound
[2019.05.06-08.52.37:645][675]LogCarla: Spawning actor 'sensor.other.collision'
[2019.05.06-08.52.38:333][677]LogCarla: FOpenDrive: Corrected map name from NewMap to NewMap
[2019.05.06-08.52.38:333][677]LogCarla: Loading OpenDrive file '../../../../carlaMap/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/NewMap.xodr'
[2019.05.06-08.52.38:334][677]LogTemp: Error: Failed to load OpenDrive file '../../../../carlaMap/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/NewMap.xodr'

XGodina, 2019-05-06:
Hi @zx627,
If you want only run with manual control you need to change the GameMode. "TheNewCarlaGameMode"
In Unreal the top tab "Blueprints" -  Project Setting - SelectGameModeBase Class and select  CarlaGameMode.

0797liyn, 2019-05-07:
@XGodina Thank you for your help.
 I can operate the vehicle with my own map. 
But when i operate the vehicle for a few seconds, a lot of vehicles fall into the map at the start point.
If not operate the vehicle when play started, it not happened. 
Is it necessary to modify same config files?

> [2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: request failed, libcurl error: 35 (SSL connect error)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 0 (Hostname in DNS cache was stale, zapped)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 1 (  Trying 52.2.113.85...)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 2 (Connected to datarouter.ol.epicgames.com (52.2.113.85) port 443 (#40))
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 3 (Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 4 (TLSv1.2 (OUT), TLS Unknown, Certificate Status (22):)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 5 (TLSv1.2 (OUT), TLS handshake, Client hello (1):)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 6 (Unknown SSL protocol error in connection to datarouter.ol.epicgames.com:443 )
[2019.05.07-08.22.03:424][650]LogHttp: Warning: 0x7f4bf591df80: libcurl info message cache 7 (Closing connection 40)
[2019.05.07-08.22.03:424][650]LogHttp: Warning: Retry exhausted on https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?SessionID=%7B29B4C780-08D6-D2FC-C373-3E3620604F55%7D&AppID=UEEditor.UnrealEngine.Release&AppVersion=4.21.2-0%2B%2B%2BUE4%2BRelease-4.21&UserID=0002460a08d6bf55659ac78a926844d1%7C%7C317e00b6a4fd4102a0d691e30ed1378a&AppEnvironment=datacollector-source&UploadType=eteventstream
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: request failed, libcurl error: 35 (SSL connect error)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 0 (Hostname in DNS cache was stale, zapped)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 1 (  Trying 18.214.253.233...)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 2 (Connected to datarouter.ol.epicgames.com (18.214.253.233) port 443 (#41))
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 3 (Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 4 (TLSv1.2 (OUT), TLS Unknown, Certificate Status (22):)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 5 (TLSv1.2 (OUT), TLS handshake, Client hello (1):)
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 6 (Unknown SSL protocol error in connection to datarouter.ol.epicgames.com:443 )
[2019.05.07-08.24.03:820][221]LogHttp: Warning: 0x7f4cb60e13c0: libcurl info message cache 7 (Closing connection 41)
[2019.05.07-08.24.03:828][222]LogHttp: Warning: Retry exhausted on https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?SessionID=%7B29B4C780-08D6-D2FC-C373-3E3620604F55%7D&AppID=UEEditor.UnrealEngine.Release&AppVersion=4.21.2-0%2B%2B%2BUE4%2BRelease-4.21&UserID=0002460a08d6bf55659ac78a926844d1%7C%7C317e00b6a4fd4102a0d691e30ed1378a&AppEnvironment=datacollector-source&UploadType=eteventstream
[2019.05.07-08.24.13:801][418]LogSlate: FSceneViewport::OnFocusLost() reason 0
[2019.05.07-08.25.11:058][285]LogCarla: Vehicle 1/80 late spawned
[2019.05.07-08.25.14:059][645]LogCarla: Vehicle 2/80 late spawned
[2019.05.07-08.25.17:062][  1]LogSpawn: Warning: SpawnActor failed because of collision at the spawn location [X=-675195.875 Y=-27800.000 Z=91816.000] for [JeepWranglerRubicon_C]
[2019.05.07-08.25.17:062][  1]LogCarla: Vehicle 2/80 late spawned
[2019.05.07-08.25.20:069][353]LogCarla: Vehicle 3/80 late spawned
[2019.05.07-08.25.23:069][710]LogCarla: Vehicle 4/80 late spawned
[2019.05.07-08.25.26:071][ 69]LogSpawn: Warning: SpawnActor failed because of collision at the spawn location [X=-675195.875 Y=-27800.000 Z=91816.000] for [VolkswagenT2_C]
[2019.05.07-08.25.26:071][ 69]LogCarla: Vehicle 4/80 late spawned
[2019.05.07-08.25.29:073][429]LogCarla: Vehicle 5/80 late spawned
[2019.05.07-08.25.32:077][787]LogCarla: Vehicle 6/80 late spawned
[2019.05.07-08.25.35:079][147]LogCarla: Vehicle 7/80 late spawned
[2019.05.07-08.25.37:280][398]LogBlueprintUserMessages: Early EndPlayMap Detection: Level '/Game/Carla/Maps/NewMap.NewMap:PersistentLevel' has LevelScriptBlueprint '/Game/Carla/Maps/NewMap.NewMap:PersistentLevel.NewMap' with GeneratedClass '/Game/Carla/Maps/NewMap.NewMap_C' with ClassGeneratedBy '/Game/Carla/Maps/NewMap.NewMap:PersistentLevel.NewMap'
[2019.05.07-08.25.37:280][398]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2019.05.07-08.25.37:386][398]LogPlayLevel: Display: Shutting down PIE online subsystems
[... cut ...]
[2019.05.07-08.26.04:783][658]LogHttp: Warning: 0x7f4bf6a03f80: libcurl info message cache 7 (Closing connection 42)
[2019.05.07-08.26.04:784][658]LogHttp: Warning: Retry exhausted on https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?SessionID=%7B29B4C780-08D6-D2FC-C373-3E3620604F55%7D&AppID=UEEditor.UnrealEngine.Release&AppVersion=4.21.2-0%2B%2B%2BUE4%2BRelease-4.21&UserID=0002460a08d6bf55659ac78a926844d1%7C%7C317e00b6a4fd4102a0d691e30ed1378a&AppEnvironment=datacollector-source&UploadType=eteventstream

0797liyn, 2019-05-09:
hi @XGodina 

- when run carla with CarlaGameMode, is it possiable to get informations of vehicle(eg,speed,location) with python script?

- if it is possiable, is there any .py examples referable?

- if it is impossiable ,Which parts of manual_control.py should i modify ,to not load .xodr?

closed 2019-07-15 by stale[bot]