carla-simulator/carla #1543: [0.9.5] automatic_control.py can not run

sheepwxy, 2019-04-16:
hello, everyone.
In Carla0.9.5,After I run the script automatic_control.py. It shows the error as below, anyone can help me? thanks!

> pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "automatic_control.py", line 88, in <module>
    from agents.navigation.roaming_agent import RoamingAgent
ImportError: No module named agents.navigation.roaming_agent

ajboloor, 2019-04-17:
You need to add carla to your `PYTHONPATH`.
Example: 
```
export PYTHONPATH=$PYTHONPATH:carla-0.9.5/PythonAPI/carla/
```

sheepwxy, 2019-04-18:
> 
> 
> You need to add carla to your `PYTHONPATH`.
> Example:
> 
> ```
> export PYTHONPATH=$PYTHONPATH:carla-0.9.5/PythonAPI/carla/
> ```

thans very much! your way helps me a lot.

closed 2019-06-24 by stale[bot]

tsc1989, 2019-09-05:
For windows this will help.
install package using pip utility as the following

go to directory PythonAPI/carla/agents
and create setup.py file and write the following code in that file

from distutils.core import setup
```
setup(name='agents',
        version='0.9.5', #doesn't matter I guess
        py_modules=['agents'],
        )
```
Now go to anaconda terminal and do cd to directory PythonAPI/carla
and apply following command to install agents

pip install -e agents

This will install agents and resolve the agents.navigation not found issue