carla-simulator/carla #1547: Strange behavior in scenario runner

deepakreddy63, 2019-04-16:
I run the same scene by adding one, two vehicles.
The scene works fine for one vehicle but shows below error when I add one or more other vehicles.

[image]

What is confusing is that It runs for two or three other vehicles once a while.

I'm not sure as to what might be causing the other vehicle to have an empty sequence when I'm clearly defining my sequence. 
P.S: It runs fine for all variations when I used Carla map but when I used a new map it has problems. Is this map related issue? If yes, Why does it run once a while in a new map?

Thanks

deepakreddy63, 2019-04-18:
This problem is because of the custom made map with sparse waypoints at that location.
It worked when I shifted to other location on the same map.

closed 2019-04-18 by deepakreddy63

YDSh510, 2019-05-06:
@ deepakreddy63

But how waypoints are generated . If I an adding new map using Road runner, will it get generated automatically for my "Map" when I run it or do I need to run some script?

deepakreddy63, 2019-05-06:
In my case, I had the open drive file (.xodr) along with the map which has the waypoint information. I think, 
You can generate this file from road runner or follow instructions in carla.org site.