carla-simulator/carla #1551: how to build carla 0.8.4 as server?

geekhut, 2019-04-18:
after going thru the tutorial and building carla 0.8.4 from source, the simulator launches in unreal editor and when I click play, it is a game for me to control a vehicle, I am not able to connect my python client to the simulator. How to build from source as a server by default?

Bobcat52, 2019-04-18:
Just run the command:
```commandline
make package
```

geekhut, 2019-04-22:
> Just run the command:
> 
> ```
> make package
> ```

sorry this doesnt run

geekhut, 2019-04-25:
somebody help..

geekhut, 2019-04-29:
Ok I managed to find the solution myself, 

https://carla.readthedocs.io/en/stable/faq/

**Can I run the server from within Unreal Editor?**
Yes, you can connect the Python client to a server running within Unreal Editor as if it was the standalone server. Go to **"Unreal/CarlaUE4/Config/CarlaSettings.ini"** (this file should have been created by the Setup.sh) and enable networking. If for whatever reason you don't have this file, just create it and add the following wzxhzdk:0 Now when you hit Play the editor will hang until a client connects.

closed 2019-07-05 by stale[bot]