carla-simulator/carla #1553: How to gain collision information ?

gut-yu, 2019-04-18:
Hello ，
I am using Carla_0.9.4 on windows10. 
Recently, I focus on motion planning problem. After one episode(sometimes when a collision occurs) , the scene must be reset.
    1、I try to read and modify the codes of manual_control.py. It is troublesome and i do not want to call too many useless classes.
    2、I know from Carla_0.9.1: Add collision event sensor, but i do not understand how it works. Could you please take some examples ? Even the simplest reaction(whether a collision occurs) is okay.  
Help!
@nsubiron @iFuSiiOnzZ

Bobcat52, 2019-04-18:
Well, it is all about callback function now, so you need to create a collision sensor with the blueprint library and then define the function (in this case, function_handler()) that will be called each time  the vehicle hits something.

```python

blueprint_library = world.get_blueprint_library()

collision_sensor = world.spawn_actor(blueprint_library.find('sensor.other.collision'),
                                        carla.Transform(), attach_to=vehicle)

collision_sensor.listen(lambda event: function_handler(event))
```

Hope it helps

nsubiron, 2019-04-18:
Like that but you added the handler for the lane invasion sensor :wink: it would be something like this
```python
def function_handler(event):
    actor_we_collide_against = event.other_actor
    impulse = event.normal_impulse
    intensity = math.sqrt(impulse.x**2 + impulse.y**2 + impulse.z**2)
```

gut-yu, 2019-04-19:
Got it !
Now i can enjoy coding.
Thanks a lot! 
@nsubiron @JMorceaux

closed 2019-04-19 by gut-yu

shuhangchen, 2020-06-16:
> Like that but you added the handler for the lane invasion sensor  it would be something like this
> 
> ```python
> def function_handler(event):
>     actor_we_collide_against = event.other_actor
>     impulse = event.normal_impulse
>     intensity = math.sqrt(impulse.x**2 + impulse.y**2 + impulse.z**2)
> ```

Hi @nsubiron , thanks for your answer along with your other replies that I benefitted from. 

   May I ask what do you suggest if we want to use collision detector in the synchronous mode?

     1. We retrieve data from the queue each frame and wait out for those frames that no collision happened?

     2. Or we only trigger an event when it happens and compare the timestamp or frame number?

Thanks!

ahmedhatem3h, 2020-07-02:
Hi @shuhangchen 
Have you find a way to use collision sensor with synchronous mode? I am also trying to make a Vehicle Class where each vehicle has it own collision sensor. When I create many instances of this vehicle class, some of the cars has their collision sensor working and some don't, and I have no idea why.

imoneoi, 2021-03-30:
@ahmedhatem3h I had the same issue, did you have any solutions?

ahmedhatem3h, 2021-03-30:
@imoneoi Yes, I used Carla 0.9.9.4, which was the latest at the time, instead of 0.9.7, and it worked fine

imoneoi, 2021-03-30:
@ahmedhatem3h Thanks! I use 0.9.11, it seems still have that issue.

TzabarDolev, 2023-09-18:
Hi,
I just wrote a short blog post about collision detectors on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/18/collision-detection-systems-in-cars-enhancing-safety-and-realism-in-carla-simulator/, and added a code example for it.

SExpert12, 2024-01-01:
> Hi, I just wrote a short blog post about collision detectors on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/18/collision-detection-systems-in-cars-enhancing-safety-and-realism-in-carla-simulator/, and added a code example for it.

Hi, You have write for one vehicle only. How to change the code for multiple vehicle in synchronous mode?
What should be location of the sensor? Like do i need to mount it on every vehicle?

Need your suggestion !!
Thanks

SExpert12, 2024-01-02:
> Like that but you added the handler for the lane invasion sensor 😉 it would be something like this
> 
> ```python
> def function_handler(event):
>     actor_we_collide_against = event.other_actor
>     impulse = event.normal_impulse
>     intensity = math.sqrt(impulse.x**2 + impulse.y**2 + impulse.z**2)
> ```

I am spawning vehicles like this:
for n, transform in enumerate(points):
            vehicle = world.spawn_actor(vehicle_bp, transform)
            collision_sensor = world.spawn_actor(collision_sensor_bp, col_transform, attach_to=vehicle, attachment_type=carla.AttachmentType.Rigid)
            vehicle.apply_control(carla.VehicleControl(throttle=0.35, steer=0))
            # batch.append(carla.command.SpawnActor(vehicle_bp, transform)

and receiving actor information through world snapshot as I am working with synchronous mode and code  is:
while start_time <= end_time:
            world.tick(10)
        
            world_snapshot = world.get_snapshot()
            for actor_snapshot in world_snapshot:
                actual_actor = world.get_actor(actor_snapshot.id)
                #print("BEFORESSSS",actual_actor,actual_actor.id)
                if actual_actor.type_id == 'vehicle.tesla.model3':

Now where to write code for collision event to get detected?
I have writing after spawning vehicle and in actor snapshot but not working.
Need help please!!!