carla-simulator/carla #1555: What is a waypoint?

TDHTTTT, 2019-04-19:
I know a waypoint is on the center of the lane. However, what is a waypoint, geometrically? How can a waypoint has rotation if it is really a point with no volume?

0102TH, 2019-04-19:
@TDHTTTT 
A waypoint is to guide vehicles to run following our scheduled track.Its algorithm is PID,so it is susceptible to interference.A waypoint can be divided into many same waypoints via our operation to make vehicles run on the track of waypoints.

TDHTTTT, 2019-04-19:
> @TDHTTTT
> A waypoint is to guide vehicles to run following our scheduled track.Its algorithm is PID,so it is susceptible to interference.A waypoint can be divided into many same waypoints via our operation to make vehicles run on the track of waypoints.

Thanks for answering. The vehicle not the waypoint is controlled by PID, right? I understand waypoints can guide vehicles but what are waypoints? Why do they have rotational information?

bernatx, 2019-04-23:
Think in that rotation as the direction to follow to keep driving in the lane. One waypoint has the direction to follow to the next waypoint on the same lane, so they have a location and a direction (a transform).

TDHTTTT, 2019-04-25:
> Think in that rotation as the direction to follow to keep driving in the lane. One waypoint has the direction to follow to the next waypoint on the same lane, so they have a location and a direction (a transform).

Thanks! Are those also defined in OpenDrive (e.g. <center> )? @bernatx

bernatx, 2019-04-25:
No, waypoints are calculated in real time and placed over the geometry of a lane at demand, so they does not exist on file.

TDHTTTT, 2019-04-25:
> No, waypoints are calculated in real time and placed over the geometry of a lane at demand, so they does not exist on file.

Thanks!

YDSh510, 2019-05-06:
But how waypoints are generated . If I an adding new map using Road runner, will it get generated automatically for my "Map" when I run it or do I need to run some script?

bernatx, 2019-05-06:
They don't need to be generated, they are calculated in real time using the open drive information. If you use Road Runner then the open drive information will be saved with your map in a file with .xodr extension, and that file is what is required to calculate the waypoints.

closed 2019-07-12 by stale[bot]