carla-simulator/carla #1556: How do you add your own dynamics model

qiujingshui, 2019-04-19:
hi,
How do you add your own dynamics model,for example carsim model

z1223343, 2019-04-19:
Similar question.

**I want to use existing powertrain and dynamics vehicle model and use Carla as a visualization tool in connected automated vehicle scenario.** 
In this case, it is expected to control the vehicle by speed or position, instead of by gas or brake.

I know Carla use PhysX model from Unreal Engine, and there are APIs to customize torque curves or tire friction parameters. But actually these are already in the existing model, I just want to make the vehicle speed profile in carla same as the model in Simulink. **Is there a way to directly control the vehicle speed without using the default vehicle model?**

Appreciated any help or suggestion!

Derekabc, 2019-05-01:
@z1223343 have you figured out it, friend?

z1223343, 2019-05-01:
@Derekabc 
Hello! Did not really work on this problem very much recently, but you may want to use `set_location(location)`, `set_transform(transform)`,  `set_velocity(vector)` for `carla.Actor` class in python client to directly control the speed or position of the vehicle. However, here may comes new problems as far as I found:
1. Need to keep the time dimension of Carla & Simulink model same as each other (1s in Carla should be same as 1s in Simulink), otherwise I think the speed control may not make sense. 
2. Currently I use TCP to communicate Carla and Simulink, and start and pause Simulink model every time step to wait for control signals from Carla and set parameter in Simulink accordingly. I found the client working in this way still too slow.

closed 2019-07-07 by stale[bot]