carla-simulator/carla #1561: How can i see all the waypoint

fty0724, 2019-04-22:
Hello!
i have already know how to modify the start and the end position in a route of my vehicle by using set_destination, but how can i customize the starting point like determining specific coordinates(x=1,z=1),instead of choosing the point in the API-spawn_waypoint[n]
Besides, is there a global top view of all the waypoint of the API?

fty0724, 2019-04-22:
As for the first question, i have see it in #1487 ,so can someone tell me is there a global top view of all the waypoint?

kartik144, 2019-04-22:
> As for the first question, i have see it in #1487 ,so can someone tell me is there a global top view of all the waypoint?

You can query all the waypoints using [generate_waypoints()](https://carla.readthedocs.io/en/latest/python_api/#carlamap) and then draw them using any of the drawing functions like `draw_point()` or `draw_string()` defined in [carla.DebugHelper](https://carla.readthedocs.io/en/latest/python_api/#carladebughelper). I personally prefer the `draw_string` function to the `draw_point` function due to it's visibility.

I suppose this code snippet might be helpful:
```
waypoints = map.generate_waypoints(distance)
for w in waypoints:
    world.world.debug.draw_string(w.transform.location, 'O', draw_shadow=False,
                                       color=carla.Color(r=255, g=0, b=0), life_time=120.0,
                                       persistent_lines=True)
```

This would draw all the waypoints at an interval of defined distance on the CarlaUE4 editor.

sheepwxy, 2019-04-23:
i try to copy your code directly except custom the (distance),but after i added codes to the automatic.py- 'game loop' part,it was warning that `AttributeError:` 'builtin_function_or_method' object has no attribute `'generate_waypoints'`,but i can see this API is exist actually in the Docs-Python API Reference,so i wonder know if i put the codes in the wrong place?
Btw,when i try to use the map.get_waypoint before,the same warning also occurred.

kartik144, 2019-04-23:
The `generate_waypoints()` function is a function of `carla.Map` class and you have to call it through it's object. If you are using the `automatic_control.py` script gievn in Python API, just modify the above code as follows:
```
waypoints = world.map.generate_waypoints(distance)
for w in waypoints:
    world.world.debug.draw_string(w.transform.location, 'O', draw_shadow=False,
                                       color=carla.Color(r=255, g=0, b=0), life_time=120.0,
                                       persistent_lines=True)
```
and paste it in the `game_loop` after these lines:
```
hud = HUD(args.width, args.height)
world = World(client.get_world(), hud)
controller = KeyboardControl(world, False)
```
and try.

sheepwxy, 2019-04-23:
Thanks for your help! I solve the problem and now I can custom the start point and destination in the specific location as I want!

sheepwxy, 2019-04-23:
Also I think he wants to see the planform which marks all the waypoints in the array of self.map.get_spawn_points(),not meaning the guided arrow of the waypoints?
Do you know how to see the full top view of the whole town?

kartik144, 2019-04-26:
Hi @sheepwxy. Sorry for the late reply.
The code I have given above
> ```
> waypoints = world.map.generate_waypoints(distance)
> for w in waypoints:
>     world.world.debug.draw_string(w.transform.location, 'O', draw_shadow=False, [...]
does exactly that. After you run the snippet, you can see the results on the CarlaUE4 simulator that pops up when you run the `CarlaUE4.sh` executable (or `CarlaUE4.exe` if you have Windows).

Once you have run the code snippet, just zoom out in the simulator (which is in spectator view) and you will be able to see all the waypoints.

sheepwxy, 2019-04-26:
Sorry,I have just seen what you meaning now! I didn't notice the UE4 interface until now
But actually I was trying to see the visible points like this,do you have any ideas?
[image](https://user-images.githubusercontent.com/31101912/56789611-def99f00-6834-11e9-8151-b77e804eb8f2.png)

Derekabc, 2019-05-05:
@sheepwxy if you can have a function returning a visible points like the image, that's cool.

Derekabc, 2019-05-05:
HI, @sheepwxy , can you give a short summary for how to set customized start / end points, that would be helpful!! THanks

sheepwxy, 2019-05-06:
@Derekabc The image is from the official website of Calla[https://carla.readthedocs.io/en/stable/benchmark_creating/](url),and I was also think it's cool but have no idea to realize it.

You can see the function set_destination in basic_agent.py, I think it may help you, then I set my points in automatic.py like  `spawn_point = self.map.get_waypoint(carla.Location(x=0,y=30,z=10))`   to define start and `agent.set_destination([100,20,4])` to define my end point

Derekabc, 2019-05-06:
@sheepwxy , thank you for your reply!  I tried you method, edit the basic_agent.py like this
`# start_waypoint = self._map.get_waypoint(self._vehicle.get_location())
        start_waypoint = self._map.get_waypoint(carla.Location(x=2.0, y=315, z=10))
        end_waypoint = self._map.get_waypoint(
            carla.Location(location[0], location[1], location[2]))`
where the location is given by the automatic.py as 
`agent.set_destination([1.8, 23, 5])`
But when i run the code, the car does not start from the point i want,  and i set the auto model, the car always collide with other things.  Any suggestion?  Thank you

fty0724, 2019-05-07:
@Derekabc I think you shouldn't change the code in basic_agent.py, since the location it  cites is from the automatic_control.py ,you can see automatic_control.py from line140 # Spawn the player and I changed the spawn point there.
Also I think because of some rules inherent ,the location may has a little offset.
If the spawn point has been occupied by buildings and the Carla won't start, you can try to change your point's position.

Derekabc, 2019-05-07:
@fty0724 Thank you! I modified the automatic_control.py.   I am using the Carla 0.9.5 and Town01.  The difference is I added `spawn_point.transform` instead of  `spawn_point` there. 
```
 while self.player is None:
            spawn_point = self.map.get_waypoint(carla.Location(x=366.54, y=327.05, z=0))
            print(spawn_point)
            self.player = self.world.try_spawn_actor(blueprint, spawn_point.transform) 
```
and set the destination as 
```
agent = BasicAgent(world.player)
agent.set_destination([212, 326, 5])
```
it fails for most of time, I would get an error:

> [2019.05.07-20.41.50:587][814]LogCarlaServer: Responding error: Spawn failed because of collision at spawn position
[2019.05.07-20.41.50:886][815]LogCarla: Spawning actor 'vehicle.bmw.grandtourer'
[2019.05.07-20.41.50:887][815]LogSpawn: Warning: SpawnActor failed because of collision at the spawn location [X=36654.000 Y=32660.998 Z=0.000] for [BmwGranTourer_C]
[2019.05.07-20.41.50:888][815]LogCarla: Warning: Failed to spawn actor 'vehicle.bmw.grandtourer'

It's really weird, if the waypoint is not accessed, but I can actually get access for that point sometimes. Any ideas?

fty0724, 2019-05-08:
@Derekabc Maybe you can try z=10,because it may stuck in the ground if z=0,just my guess.
Also I met this error before, and I changed the point and sometimes it succeeded.

Derekabc, 2019-05-08:
does you car achieve its destination on your setting?  my car cannot stop at the destination point and keeps going forever.

fty0724, 2019-05-08:
My car can stop when it arrive around the destination i set. And I think my code is same as yours since it only needs to change agent.set_destination.Sorry that I never met your situation before.

Derekabc, 2019-05-08:
I plan to try different maps. I used to run my car on Town01 and Carla 0.9.5. However, it almost failed. BTW, would like to share your setting of carla, such as location for starting /end points and version of carla, so i can have a try and find the problem.

sheepwxy, 2019-05-10:
@Derekabc sorry to reply you so late here is my setting of carla
`# Spawn the player.
        if self.player is not None:
            spawn_point = self.player.get_transform()
            spawn_point.location.z += 2.0
            spawn_point.rotation.roll = 0.0
            spawn_point.rotation.pitch = 0.0
            self.destroy()
            self.player = self.world.try_spawn_actor(blueprint, waypoint)
        while self.player is None:
            spawn_points = self.map.get_spawn_points()
	    spawn_point = self.map.get_waypoint(carla.Location(x=0,y=30,z=10))
            self.player = self.world.try_spawn_actor(blueprint,spawn_point.transform)`

and in the game loop
`try:
        client = carla.Client(args.host, args.port)
        client.set_timeout(4.0)

        display = pygame.display.set_mode(
            (args.width, args.height),
            pygame.HWSURFACE | pygame.DOUBLEBUF)

        hud = HUD(args.width, args.height)
        world = World(client.get_world(), hud, args.filter)
        controller = KeyboardControl(world, False)
        if args.agent == "Roaming":
            agent = RoamingAgent(world.player,target_speed=20)
        else:
            agent = BasicAgent(world.player,target_speed=20)
	    agent.set_destination([100,20,4])`

closed 2019-07-16 by stale[bot]