carla-simulator/carla #1567: Carla on ppc64le building of Unreal Engine fails

robertengelmann, 2019-04-24:
Hi everyone,

I am trying to build Carla for the `ppc64le` architecture on a POWER8 machine. The machine runs RHEL 7.5 and I am trying to build in a Docker container based on an NVIDIA CUDA Ubuntu 16.04 image. So far I have basically followed the documentation for building on Linux (<https://carla.readthedocs.io/en/latest/how_to_build_on_linux/>), but when compiling the Unreal Engine, I get two error messages I have not been able to fix, yet:
When executing `Setup.sh` it exits successfully, but I get:  
 `Register the engine installation...  
./Setup.sh: line 45: ./UnrealVersionSelector-Linux-Shipping: cannot execute binary file: Exec format error`  
Looks to me like this binary is not compiled for ppc64le, but I don't even know if it is important.   When running `GenerateProjectFiles.sh`:  
`/usr/lib/mono/4.5/Microsoft.CSharp.targets: error : Compiler crashed with code: 139.
Cannot open assembly './../../../Binaries/DotNET/UnrealBuildTool.exe': No such file or directory.`  
I have checked, the file `UnrealBuildTool.exe` is really not there. I suppose it should have been generated in a previous step.  
I have found this pull request to the UE project <https://github.com/EpicGames/UnrealEngine/pull/2585>, but have not been able to successfully incorporate it.  

Has anyone tried this before? Any experiences or ideas to share? Am I missing something?  
Please ask, if I have left anything obscured. I have also posted this in the Unreal Engine forums.

Any help would be appreciated :-)

klauspower, 2019-04-29:
i think the dll is 32bit .. (must change mono ) and dll  thx klaus

klauspower, 2019-04-29:
check: > https://www.mono-project.com/archived/installerinstructions/
Will the installer work on x86_64?

No, this installer is only built to run on 32 bit systems. Some distributions have both 32 bit and 64 bit libraries and programs installed. If a 64 bit system has all the necessary 32 bit components installed, it should work, although this is not a tested scenario.

so .. power8 i a native 64bit system .. 
ps. will try it on power9 ..  ()  I don’t know that for sure yet !

klauspower, 2019-04-29:
look .. yes .. 
must build an VM .. 
https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html-single/7.6_release_notes/index

[1] Note that the Red Hat Enterprise Linux 7.6 installation is supported only on 64-bit hardware. Red Hat Enterprise Linux 7.6 is able to run 32-bit operating systems, including previous versions of Red Hat Enterprise Linux, as virtual machines.

robertengelmann, 2019-04-30:
Thank you for your input, @klauspower . When you say
> i think the dll is 32bit .. (must change mono ) and dll
do you talk about `./UnrealVersionSelector-Linux-Shipping` ?
Are you suggesting the Unreal Engine can only be built on 32 bit systems?

klauspower, 2019-05-03:
no but x86 can both 64 and 32bit ..  power only 64-bit --
mono  is only 32-bit .. 
to. docker .. https://forums.docker.com/t/support-for-32-bit-images-containers-on-64-bit-hosts/10010
to vm  .. need  kvm ! 
understand .. 
try to build () UE4 on x86 .. after make it on power8 / 9  you must understand mono .. 
no native .. need docker or vm and try ..  
ps. will try it on power9  ( )I'll let you know when it's ready.

closed 2019-07-09 by stale[bot]

edowson, 2019-08-19:
I'm porting Raptor Engineering's port of UE4.20.3-release from the [kb9vqf/UnrealEngine/tree/release](https://github.com/kb9vqf/UnrealEngine/tree/release) branch.

You can track it's progress in the [edowson/UnrealEngine/tree/4.22-ppc64le](https://github.com/edowson/UnrealEngine/tree/4.22-ppc64le) branch.

Additional notes for setting up and installing Ubuntu-18.04 on a Power9 system like the Talos II Secure Workstation can be found in the [edowson/raptor-talos-ii-power9-technote](https://github.com/edowson/raptor-talos-ii-power9-technote) repo. It includes notes for setting NVIDIA Tesla GPU drivers and CUDA.

I still have to port several fixes for third party libraries to `ppc64el`, it might take me another week or so to finish this.

In the meantime, you can prepare a Power9 host using the technotes, as well as install the following packages on your system, in order to get the a basic environment to compile the UE4.22.3 sources on `ppc64le`

Install common development packages:
```bash
sudo apt install --no-install-recommends -q -y \
python-dev \
python3-dev

sudo apt-get install --no-install-recommends -q -y \
autotools-dev \
autoconf \
automake \
cmake \
dos2unix \
libtool

sudo apt-get install --no-install-recommends -q -y \
libtool libc++-dev libc++abi-dev \
libglib2.0-dev libhdf5-dev libwebsockets-dev \
freeglut3-dev  libvulkan-dev wayland-protocols \
libegl1-mesa-dev libglu1-mesa-dev libgl1-mesa-dev libgles2-mesa-dev mesa-common-dev
```

Install required libraries:
```bash
sudo apt-get install --no-install-recommends -q -y \
libfreetype6-dev \
libomp-dev
```

Install X11 development libraries:
```bash
sudo apt-get install --no-install-recommends -q -y \
libx11-dev \
libxcursor-dev \
libxext-dev \
libxi-dev \
libxt-dev \
libxv-dev \
libxinerama-dev \
libxkbcommon-dev \
libxrandr-dev \
libxrender-dev \
x11-xkb-utils \
x11proto-input-dev \
x11proto-randr-dev \
x11proto-scrnsaver-dev \
x11proto-xinerama-dev \
x11-xserver-utils \
xauth \
xdg-user-dirs \
xdg-utils \
xfonts-base \
xkb-data \
```

Install mono development tools:
```bash
# setup sources.list
sudo apt install gnupg ca-certificates
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
echo "deb https://download.mono-project.com/repo/ubuntu stable-bionic main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update

sudo apt-get install --no-install-recommends -q -y \
mono-devel \
mono-dmcs
```

Build clang-8.0.1 on linux by running the [Engine/Build/BatchFiles/Linux/Toolchain/BuildClangOnLinux.sh](https://github.com/edowson/UnrealEngine/blob/4.22-ppc64le/Engine/Build/BatchFiles/Linux/Toolchain/BuildClangOnLinux.sh) script.

qhaas, 2019-11-06:
Thanks, @edowson .  I've been monitoring your work on your 4.22 ppc64le form of ue4.  Is the above your current development environment?  If you have it compiling, a text file with the results of `apt list --installed` would be quite useful to us.

edowson, 2019-11-07:
@qhaas The latest set of patches can be found in the [edowson/UnrealEngine/commits/4.23-ppc64le](https://github.com/edowson/UnrealEngine/commits/4.23-ppc64le) branch.

edowson, 2019-11-07:
The output of `apt list --installed` on the power9 system is as follows:

```
accountsservice/bionic,now 0.6.45-1ubuntu1 ppc64el [installed]
acl/bionic,now 2.2.52-3build1 ppc64el [installed]
acpid/bionic,now 1:2.0.28-1ubuntu1 ppc64el [installed]
activity-log-manager/bionic,now 0.9.7-0ubuntu26 ppc64el [installed,automatic]
adduser/bionic,now 3.116ubuntu1 all [installed]
adium-theme-ubuntu/bionic,now 0.3.4-0ubuntu4 all [installed]
adwaita-icon-theme/bionic,now 3.28.0-1ubuntu1 all [installed]
aisleriot/bionic,now 1:3.22.5-1 ppc64el [installed]
alsa-base/bionic,now 1.0.25+dfsg-0ubuntu5 all [installed]
alsa-utils/bionic,now 1.1.3-1ubuntu1 ppc64el [installed]
anacron/bionic,now 2.3-24 ppc64el [installed]
apg/bionic,now 2.2.3.dfsg.1-5 ppc64el [installed]
app-install-data-partner/bionic,now 16.04 all [installed]
apparmor/bionic-updates,bionic-security,now 2.12-4ubuntu5.1 ppc64el [installed]
apport/now 2.20.9-0ubuntu7.7 all [installed,upgradable to: 2.20.9-0ubuntu7.9]
apport-gtk/now 2.20.9-0ubuntu7.7 all [installed,upgradable to: 2.20.9-0ubuntu7.9]
apport-symptoms/bionic,now 0.20 all [installed]
appstream/bionic-updates,now 0.12.0-3ubuntu1 ppc64el [installed]
apt/bionic-updates,now 1.6.12 ppc64el [installed]
apt-config-icons/bionic-updates,now 0.12.0-3ubuntu1 all [installed]
apt-transport-https/bionic-updates,now 1.6.12 all [installed]
apt-utils/bionic-updates,now 1.6.12 ppc64el [installed]
aptdaemon/bionic-updates,now 1.1.1+bzr982-0ubuntu19.1 all [installed]
aptdaemon-data/bionic-updates,now 1.1.1+bzr982-0ubuntu19.1 all [installed]
apturl/bionic-updates,now 0.5.2ubuntu14.2 ppc64el [installed]
apturl-common/bionic-updates,now 0.5.2ubuntu14.2 ppc64el [installed]
aspell/bionic-updates,bionic-security,now 0.60.7~20110707-4ubuntu0.1 ppc64el [installed]
aspell-en/bionic,now 2017.08.24-0-0.1 all [installed]
at/bionic,now 3.1.20-3.1ubuntu2 ppc64el [installed]
at-spi2-core/bionic,now 2.28.0-1 ppc64el [installed]
autoconf/bionic,now 2.69-11 all [installed]
automake/bionic,now 1:1.15.1-3ubuntu2 all [installed]
autotools-dev/bionic,now 20180224.1 all [installed]
avahi-autoipd/bionic-updates,bionic-security,now 0.7-3.1ubuntu1.2 ppc64el [installed]
avahi-daemon/bionic-updates,bionic-security,now 0.7-3.1ubuntu1.2 ppc64el [installed]
avahi-utils/bionic-updates,bionic-security,now 0.7-3.1ubuntu1.2 ppc64el [installed]
baobab/bionic,now 3.28.0-1 ppc64el [installed]
base-files/bionic-updates,now 10.1ubuntu2.7 ppc64el [installed]
base-passwd/bionic,now 3.5.44 ppc64el [installed,automatic]
bash/bionic-updates,now 4.4.18-2ubuntu1.2 ppc64el [installed]
bash-completion/bionic,now 1:2.8-1ubuntu1 all [installed]
bc/bionic,now 1.07.1-2 ppc64el [installed]
bcache-tools/bionic,now 1.0.8-2build1 ppc64el [installed]
bind9-host/bionic-updates,now 1:9.11.3+dfsg-1ubuntu1.9 ppc64el [installed]
binfmt-support/bionic,now 2.1.8-2 ppc64el [installed,automatic]
binutils/bionic-updates,bionic-security,now 2.30-21ubuntu1~18.04.2 ppc64el [installed]
binutils-common/bionic-updates,bionic-security,now 2.30-21ubuntu1~18.04.2 ppc64el [installed]
binutils-powerpc64le-linux-gnu/bionic-updates,bionic-security,now 2.30-21ubuntu1~18.04.2 ppc64el [installed]
bison/bionic,now 2:3.0.4.dfsg-1build1 ppc64el [installed]
bluez/now 5.48-0ubuntu3.1 ppc64el [installed,upgradable to: 5.48-0ubuntu3.2]
bluez-cups/now 5.48-0ubuntu3.1 ppc64el [installed,upgradable to: 5.48-0ubuntu3.2]
bluez-obexd/now 5.48-0ubuntu3.1 ppc64el [installed,upgradable to: 5.48-0ubuntu3.2]
bolt/bionic-updates,now 0.5-0ubuntu0.18.04.1 ppc64el [installed]
branding-ubuntu/bionic,now 0.10 all [installed]
brltty/bionic-updates,now 5.5-4ubuntu2.0.1 ppc64el [installed]
bsdmainutils/bionic,now 11.1.2ubuntu1 ppc64el [installed]
bsdutils/bionic-updates,now 1:2.31.1-0.4ubuntu3.4 ppc64el [installed]
btrfs-progs/bionic,now 4.15.1-1build1 ppc64el [installed]
btrfs-tools/bionic,now 4.15.1-1build1 ppc64el [installed]
bubblewrap/bionic-updates,bionic-security,now 0.2.1-1ubuntu0.1 ppc64el [installed,automatic]
build-essential/bionic,now 12.4ubuntu1 ppc64el [installed]
busybox-initramfs/bionic-updates,bionic-security,now 1:1.27.2-2ubuntu3.2 ppc64el [installed]
busybox-static/bionic-updates,bionic-security,now 1:1.27.2-2ubuntu3.2 ppc64el [installed]
byobu/bionic,now 5.125-0ubuntu1 all [installed]
bzip2/bionic-updates,bionic-security,now 1.0.6-8.1ubuntu0.2 ppc64el [installed]
ca-certificates/bionic,bionic-updates,now 20180409 all [installed]
ca-certificates-java/bionic-updates,now 20180516ubuntu1~18.04.1 all [installed,automatic]
[... 2036 lines cut ...]
xserver-xorg-input-all/bionic-updates,now 1:7.7+19ubuntu7.1 ppc64el [installed]
xserver-xorg-input-libinput/bionic,now 0.27.1-1 ppc64el [installed]
xserver-xorg-input-wacom/bionic,now 1:0.36.1-0ubuntu1 ppc64el [installed]
xserver-xorg-legacy/bionic-updates,now 2:1.19.6-1ubuntu4.3 ppc64el [installed]
xserver-xorg-video-all/bionic-updates,now 1:7.7+19ubuntu7.1 ppc64el [installed]
xserver-xorg-video-amdgpu/bionic,now 18.0.1-1 ppc64el [installed]
xserver-xorg-video-ati/bionic,now 1:18.0.1-1 ppc64el [installed]
xserver-xorg-video-fbdev/bionic,now 1:0.4.4-1build6 ppc64el [installed]
xserver-xorg-video-nvidia-418/unknown,now 418.87.01-0ubuntu1 ppc64el [installed,automatic]
xserver-xorg-video-radeon/bionic,now 1:18.0.1-1 ppc64el [installed]
xserver-xorg-video-vesa/bionic,now 1:2.3.4-1build3 ppc64el [installed]
xtrans-dev/bionic,now 1.3.5-1 all [installed,automatic]
xul-ext-ubufox/bionic,now 3.4-0ubuntu1.17.10.1 all [installed]
[... cut ...]
xxd/bionic-updates,bionic-security,now 2:8.0.1453-1ubuntu1.1 ppc64el [installed,automatic]
xz-utils/bionic,now 5.2.2-1.3 ppc64el [installed]
yelp/bionic,now 3.26.0-1ubuntu2 ppc64el [installed]
yelp-xsl/bionic,now 3.20.1-4 all [installed]
zeitgeist-core/bionic,now 1.0-0.1ubuntu1 ppc64el [installed]
zenity/bionic,now 3.28.1-1 ppc64el [installed]
zenity-common/bionic,now 3.28.1-1 all [installed]
zip/bionic,now 3.0-11build1 ppc64el [installed]
zlib1g/bionic,now 1:1.2.11.dfsg-0ubuntu2 ppc64el [installed]
zlib1g-dev/bionic,now 1:1.2.11.dfsg-0ubuntu2 ppc64el [installed,automatic]
```

edowson, 2019-11-07:
@qhaas The current status is that only a few more libraries need to be added to the [Engine/Build/BatchFiles/Linux/BuildThirdParty.sh](https://github.com/edowson/UnrealEngine/blob/4.23-ppc64le/Engine/Build/BatchFiles/Linux/BuildThirdParty.sh) build script, and to fix some linker errors on ppc64le.
```
KissFFT, Steam API, WebRTC
Opus
libStrophe
FBX: some more to be disabled
```

Run this script to build the ppc64le toolchain.
[Engine/Build/BatchFiles/Linux/Toolchain/DockerOnWindows/build_linux_toolchain/build_linux_toolchain.sh](https://github.com/edowson/UnrealEngine/blob/4.23-ppc64le/Engine/Build/BatchFiles/Linux/Toolchain/DockerOnWindows/build_linux_toolchain/build_linux_toolchain.sh)

Just modify line 27 to build just one targets to save time:
```
TARGETS="powerpc64le-unknown-linux-gnu"
```

The BuildThirdParty.sh has detection logic for the clang toolchain. An earlier set of patches had that for the Linux/Setup.sh script, which I will have to reintroduce. This way, if a user runs Setup.sh, it will check for existence of the clang-8 toolchain and trigger a build. I'll parameterize the build_linux_toolchain.sh script later, to accept an argument to over-ride the default target list, to save time during the build process.

qhaas, 2019-11-07:
Thanks, @edowson .  We were barking up the wrong branch (4.22-ppc64le), it would seem.  We will focus on getting your 4.23-ppc64le branch to compile on our end.  We had gotten to the point where we ran into the LLVM bug report you submitted (bug report 43072) when compiling the Epic's SDL2 fork and I backported the fix from LLVM's master branch to the 8.0.0 version that ships with Ubuntu 18.04 ppc64le.  But, you might have already worked around that issue in your 4.23 branch.  Thanks for the tip on using the toolchain script in the DockerOnWindows folder, we will try that.

Our end goal is to get CARLA running on a ppc64le / Tesla V100 based HPC in a RHEL7 environment, so it is doubtful we need the full UE4 feature-set.  We have a Raptor Talos 2 for development purposes, and we are using containers to replicate your Ubuntu 18.04 development environment.  We will likely build an Ubuntu 18.04 ppc64le runtime container on the HPC since Singularity is available.

edowson, 2019-11-07:
#qhaas The early tests were performed using a Raptor system with a GV100 GPU. You don't necessarily have to have a GV100 for these tests, an RTX2070 GPU will also work fine.

Send me a pm with your email, and we can sync our efforts directly.

I've containerized UE4 for x86_64 and did an experimental trial running it in a Kubernetes cluster. So, my end-goal is running UE4 on containers, as well.

edowson, 2019-11-07:
@qhaas There is a pull request for general script fixes here:
https://github.com/EpicGames/UnrealEngine/pull/6279

Hopefully, if that gets accepted into the master branch, then the delta ppc64le patches for future version 4.24 and above, will be much smaller.

qhaas, 2019-11-07:
@edowson , since GitHub doesn't seem to have private messages anymore, I sent an email to your gmail account that was listed in the commit logs.

qhaas, 2020-09-29:
Using @edowson 's 4.23.1-ppc64le fork mentioned above, we now have CARLA 0.9.8 running on our IBM AC922 HPC, also known as the [Summit Supercomputer](https://www.olcf.ornl.gov/summit/) and our Raptor Talos 2 workstations!  The latter was easier, for supercomputers have their own nuances...

Unfortunately, a few textures are missing (note fewer details of the plants on the right with ppc64le), likely because Epic doesn't have fallbacks for some x86-64 SIMD calls (based on errors in the logs, snippet below).

The first known image rendered with CARLA 0.9.8 on Summit (Power9+Tesla V100) using Dowson's 4.23.1-ppc64le UE fork and a slightly modified tutorial.py script:
[image]
x86-64 + Kepler era GPU using Epic's official UE 4.23.1 for comparison:
[image]
```
LogMeshUtilities: Error: Couldn't generate distance field for mesh, platform is missing required Vector intrinsics.
```

We are having to run it via the UnrealEditor due to issues with `make package`, but these issues are occurring on Epic's 4.23.1 release on x86-64 as well, so likely an artifact of using a version of CARLA (and its assets) meant for UE 4.22 with UE 4.23.1:
```
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Array ' pin is invalid: Array inputs (like 'Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Array ' pin is invalid: Array inputs (like 'Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
```

@germanros1987 , you were correct about how much time it would take when we mentioned this effort to you last year on a teleconference call, ha.

qhaas, 2020-12-02:
We now have CARLA 0.9.8 packaging in 'Shipping' mode with the unofficial ppc64le UnrealEngine 4.23.1 fork and clang-8.

The `Error: Couldn't generate distance field for mesh` was fixed with [this commit to the fork](https://github.com/edowson/UnrealEngine/commit/e3e79d00646761028d689f6a39212f97a382a2df).

Unfortunately, while we can package and run CARLA now on pp64le, the origin of the visual artifacts / lack of detail on the trees/shrubs depicted above remain elusive.  They still occur even with said fixes.  We also cannot force the issue on x86-64, even with 'low' detail settings.  There are no warnings/errors to indicate what the problem is that we have discovered.  We are assuming a driver bug at this point.

qhaas, 2021-02-21:
[Per a similar issuing occurring on ARM](https://github.com/rdeioris/UnrealOnRPI4#step-51-what-happened-to-my-textures), DXTC is apparently misbehaving on ppc64le.  When we disabled it (per the instructions in the guide linked), the artifacts / missing textures appear resolved.

With DXTC on ppc64le:
[image]
Without DXTC on ppc64le:
[image]

This was the last of the major outstanding issues with ppc64le!