carla-simulator/carla #1572: How to create a map step by step?

Derekabc, 2019-04-26:
Hello, I am new to Carla. Now I want to create a new map by RoadRunner and Unreal Engine 4. 
I really appreciate there are so many contributors to this topic. I mainly focus on these two tutorials for building my own map. 
Guide#1: [https://github.com/carla-simulator/carla/issues/1015](url) and 
Guide#2: [https://carla.readthedocs.io/en/latest/how_to_make_a_new_map](url)
I find it's pretty hard to follow these instructions to build my own map. One reason is that I do not know how to check each step is correct or not so it also takes me a lot of time to figure out this. There are my steps and questions:
1. created a map on RoadRunner. Firstly, I changed the Rotation to 180 degree followed by guide#2. Then drew my map, just a simple one. **I do not know what is OpenDRIVE Preview Tool, does this step help?** I simply created my map and exported as fbx and opendrive files. Saved using same name, test1.fbx and test1.xodr.   
2. I copied the opendrive file to `/home/dong/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/test1.xodr`. 
3. open the unreal engine and create a new level and saved it to `Maps/test`. Then imported the fbx file to a newly created folder but I got some warinings. Changed the roation to (0, 0, 0).
[image]
I got the RoadRunner plugins installed. I do not know why I cannot see the results just like this video shows. [https://www.youtube.com/watch?v=cu_IwFZhvhI](url), I need to drag the meshes to the levele by myself.  
[image]
3. I just continued to create my own map.  I dragged all the items to the level and deleted the floor in the newly created level. Then I searched the OpenDrive actor / CarlaMapGenerator and dragged them to the level.  Check "generate routes"  and then get 
[image]
I found it's difficult to move the route to the road I generated. Later I tried another created road by the same way, I found I cannot see the route anymore after clicking "generate routes" .  I want to know which step may I make mistakes? 
[image]
Thank you for your help!!

Derekabc, 2019-04-30:
@XGodina hi, I still need your help. thanks

eloy96vg, 2019-05-02:
@Derekabc i have used generate_map.py as they explain on this page (https://github.com/carla-simulator/carla/blob/master/Docs/generate_map_from_fbx.md) and have done the whole process without doing anything. It is easier than doing the manual process.

Derekabc, 2019-05-02:
@eloy96vg Thank you for your reply !  In this method, can we add new spawn points and path?

aisichongzi007, 2019-05-07:
hi , how to get  the  key  of  roadrunner.   it,s  expensive.
or ,do you have the crack of roadrunner

hieunguyen88, 2019-06-03:
Hi @Derekabc , how do you generate the route manually? I drag OpenDriveActor and CarlaMapGenerator, but I didn't see any route show up. You mentioned "clicking 'generate routes'", is there a button in UnrealEditor? I couldn't find it.

About your issue, I think the map need at least one junction (I run into the same issue when my map only has one straight road. It works when there is at least one junction). Try to make one junction in road runner and import it again

athreyasridharan, 2019-06-07:
@hieunguyen88 @Derekabc I'm also looking for the generate routes button. Could you point me to it?

Derekabc, 2019-06-07:
hi, @athreyasridharan, you see the last picture of my first comment. You can find the generate route button when you choose the CarlaMapGenerate label.

RehabHsn, 2019-07-26:
@eloy96vg  the link is not working 
could you please update the link

mehmor, 2019-08-05:
> Hello, I am new to Carla. Now I want to create a new map by RoadRunner and Unreal Engine 4.
> I really appreciate there are so many contributors to this topic. I mainly focus on these two tutorials for building my own map.
> Guide#1: [https://github.com/carla-simulator/carla/issues/1015](url) and
> Guide#2: [https://carla.readthedocs.io/en/latest/how_to_make_a_new_map](url)
> I find it's pretty hard to follow these instructions to build my own map. One reason is that I do not know how to check each step is correct or not so it also takes me a lot of time to figure out this. There are my steps and questions:
> 
> 1. created a map on RoadRunner. Firstly, I changed the Rotation to 180 degree followed by guide#2. Then drew my map, just a simple one. **I do not know what is OpenDRIVE Preview Tool, does this step help?** I simply created my map and exported as fbx and opendrive files. Saved using same name, test1.fbx and test1.xodr.
> 2. I copied the opendrive file to `/home/dong/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/test1.xodr`.
> 3. open the unreal engine and create a new level and saved it to `Maps/test`. Then imported the fbx file to a newly created folder but I got some warinings. Changed the roation to (0, 0, 0).
>    [image]
>    I got the RoadRunner plugins installed. I do not know why I cannot see the results just like this video shows. [https://www.youtube.com/watch?v=cu_IwFZhvhI](url), I need to drag the meshes to the levele by myself.
>    [image]
> 4. I just continued to create my own map.  I dragged all the items to the level and deleted the floor in the newly created level. Then I searched the OpenDrive actor / CarlaMapGenerator and dragged them to the level.  Check "generate routes"  and then get
>    [image]
>    I found it's difficult to move the route to the road I generated. Later I tried another created road by the same way, I found I cannot see the route anymore after clicking "generate routes" .
>    Also a log information `LogCarla: Loading OpenDrive file '../../../../carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/test1.xodr' WARNING: cannot parse georeference: ''. Using default values. `
>    I want to know which step may I make mistakes?
>    [image]
>    Thank you for your help!!

What if we don't have Roadrunner? what is the best option?

ndeshp2s, 2019-08-21:
Hello,

I followed the guide mentioned here [here](https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/)
to create my custom environment map. I did not get any error, however, after importing the .fbx file into unreal engine the objects in the environment have no textures. 

Why are not the material textures/colors visible in unreal? 

Road Runner map:
[image]

Unreal engine map:
[image]
[image]

Kingsman1995, 2019-10-08:
Hello, i want to ask where did u get the RoadRunner install file? Because i've contacted the Vectorzero and gotten no response yet.

closed 2019-12-14 by stale[bot]

GeraldGuo7, 2021-05-10:
It looks like Carla has some connection with Mathworks and does anyone find a way to use other software to generate .fbx and .xodr files?
If I just want to use a simple map which does not have road standards, can I just export the .fbx file to do the simulation?
Thanks in advance.