carla-simulator/carla #1580: Low fps on carla 0.9.5 when cpu, gpu and memory usage is low

user025, 2019-04-28:
Here is a quick guess: 
   do sensors of Carla generate data one by one in one thread? 
   If it is, any solution to improve it when one thread is too crowded?
   If it is not, how to get carla use full resources of my system?

I'm working on Carla 0.9.5. When I added two 16-channel lidar sensor, 288000 points per second,
the FPS fell dramatically, from 100 to 2, while GPU usage is only 50%, and CPU usage slightly over 100%(the maximum is supposed to be 32). I've tried on some other instances.  Better GPU or more CPU cores didn't help.  

I checked the code and it seemed that sensors were ticked one by one and generated their data in a synchronization way.  When time-consuming sensors are added, single cpu thread becomes the bottleneck. There won't be much improvement unless this sensor work mode is changed.

I'm not sure about the conclusion. Would you kindly offer some suggestions?

chankim, 2019-05-10:
the same here. I installed 0.9.5 yesterday and ran the exmple, and found server 1 fps, client 2 fps.

user025, 2019-05-16:
@chankim if you run example slowly, check your GPU and CPU usage using nvidia-smi and top first. Or just try on a better GPU and see if it helps.

Xiaoyu006, 2019-05-17:
I got about 10 fps, better but not enough.
Have you fixed this?
I am using 2 GPUs GTX 1080 with driver 410
and double Xeon CPU with totally 40 threads.

user025, 2019-05-21:
@Xiaoyu006  If you just want to get a point cloud for planning and control you can use a deep-view camera to mimic it, as #779 suggested.
If you focused on sensor itself, you may try following steps:

- Reduce the points per second (A safe downgrade, if downgrade is acceptable)

**The following steps are not thoroughly tested. Sidebacks are unknown and may only work for carla's sensor part**

- Rewrite to collision function in UE4 and lidar part in Carla, just lock once per-channel instead of lock every point. (Could gain a notable speed up)

- Try patch some multithread libs into The UE4's collision function.

If you are really, really interested in the last two step, maybe we can discuss it via email.

closed 2019-07-27 by stale[bot]