carla-simulator/carla #1581: Carla got stuck when spawning a collision sensor

gy20073, 2019-04-29:
carla version: 0.9.5
os: 16.04
GPU: V100, driver, 396.26

[image]

When I run Carla consecutively for a few hours (1hour - 3hour) I will got stuck when trying to spawning a collision sensor ('sensor.other.collision'). But right before spawning the collision sensor, I am able to spawn a vehicle. When getting stuck, the carla keep printing the message of "Task graph took xx ms for RT to receive broadcast", as shown in the yellow text above. 

This issue randomly appears, in the sense that sometimes it will not get stuck at that position. When it does not get stuck, it will print out the following logging message:
[image]

Derekabc, 2019-07-03:
Hi @gy20073 , have you solved this problem. I got the same problem. The server will not get stuck after hundreds of episodes of training. The FPS of client drops to 0. I also create an issue here https://github.com/carla-simulator/carla/issues/1840. Thank you!!

germanros1987, 2020-03-19:
@bernatx what do we know about this issue?

germanros1987, 2020-06-26:
@gy20073 have you seen this problem in most recent versions (0.9.9?)

gy20073, 2020-06-26:
> @gy20073 have you seen this problem in most recent versions (0.9.9?)

I haven't try the most recent version yet, not sure whether it happens

bernatx, 2020-07-02:
Currently we have not detected that problem, but if the frame rate is low the Unreal can show those yellow warning meaning that by some reason some resource is taking more time than expected. Did you try using synchronous mode?

closed 2025-04-01 by Blyron