carla-simulator/carla #1582: Make Launch on Windows error

jcrevoisier, 2019-04-29:
Get the following error when 'make launch'
Though, I checked and the  C:/Carla/carla/Util/BuildTools\Windows.mk.help is existing.
I don't understand why it is not founding it then, any idea ?

C:\Carla\carla>make launch
/usr/bin/bash: line 0: type: C:/Carla/carla/Util/BuildTools\Windows.mk.help: not found
mingw32-make.exe: *** [Util/BuildTools/Windows.mk:12: help] Error 1

jcrevoisier, 2019-04-30:
[image]

marcgpuig, 2019-04-30:
Hi @jcrevoisier.
Take a look at you line :`/usr/bin/bash:`. It is possible that you have some type of bash interpreter set on your PATH?

jcrevoisier, 2019-05-01:
Yes, Thank you !
This issue is solved, though I still have an issue when launching 'make setup' about b2.exe apparently :/ 

[image]

rajesh8646, 2019-05-01:
Hi, While running 'make lunch' - I am also getting same error. I am not understanding, how to fix it.
Machine: Windows 10  64 bit 
Using - Carla 0.9.5 
referred : https://carla.readthedocs.io/en/latest/how_to_build_on_windows/

 :...updated 14754 targets...

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

closed 2019-07-08 by stale[bot]

hitman996, 2019-12-11:
Hi @jcrevoisier , 
Could you explain how did you fix this problem? I am new to this. 

Thank you.

SIlvaMFPedro, 2021-04-20:
Hello. I am having this same issue in my Windows Build
[image]
I tried to perform a make clean but it doesn't work because it says that the Carla package is not set.
[image]
Can you explain to me how to solve this issue?
Thank you.