carla-simulator/carla #1583: [0.9.5] make launch error, Possess overrides final function

dermirt, 2019-04-29:
I am trying to launch Carla on Ubuntu. I followed all the steps in the documentation, but anyway the build process fails. After executing `make launch` or `make rebuild` the process is running for quite a while, but crashes after several errors with 

```[12/15] Compile CarlaUE4GameModeBase.gen.cpp
2 warnings and 5 errors generated.
Makefile:215: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/htwcar-master/HTWCAR/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:28: recipe for target 'rebuild' failed
make: *** [rebuild] Error 2
```
most of the errors are related to `error: declaration of 'Possess' overrides a 'final' function` or `constexpr variable cannot have non-literal type 'const`

nsubiron, 2019-04-30:
It seems you're trying to compile with UE4.22, we're currently using 4.21. See [How to build on Linux](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/).

guochun7648, 2019-05-02:
hi,
I have the same issue on make launch. Did you solve it? Please let me know. Thank you!

asvnphanindra, 2019-05-07:
Hi,

I also have a similar issue. I am currently setting up Carla at my PC. As mentioned in the documentation at Build Carla section, when I execute the command  "make launch" I get this error. Below mentioned is the detailed output. Kindly help me in solving this issue.

asvnphanindra@tesla:~/carla$ make launch
Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
................
------------------------------
Performing 22 actions (10 in parallel)
[1/22] Compile PCH.CarlaUE4.h
[3/22] Compile Module.OnlineSubsystemUtils.cpp
[5/22] Compile Module.Carla.gen.3_of_4.cpp
[2/22] Compile Module.Voice.cpp
[4/22] Compile Module.OnlineSubsystemUtils.gen.cpp
[6/22] Compile Module.Carla.gen.2_of_4.cpp
[7/22] Compile Module.Carla.gen.4_of_4.cpp
[9/22] Compile Module.Carla.cpp
[8/22] Compile Module.Carla.gen.1_of_4.cpp
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:2:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.cpp:8:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h:13:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/VehicleParameters.h:9:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.h:13:
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:2### **8:18: error: constexpr variable cannot have non-literal type 'const (lambda at /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:28:25)'###
  constexpr auto Cast = [](auto e) { return static_cast<typename std::underlying_type<decltype(e)>::type>(e); };
                 ^
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:28:25: note: lambda closure types are non-literal types before C++17
  constexpr auto Cast = [](auto e) { return static_cast<typename std::underlying_type<decltype(e)>::type>(e); };
                        ^
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.4_of_4.cpp:3:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/WalkerController.gen.cpp:8:
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.h:25:8: error: declaration of 'Possess' overrides a 'final' function
  void Possess(APawn *InPawn) override;
       ^
/home/asvnphanindra/UnrealEngine/UnrealEngine/Engine/Source/Runtime/Engine/Classes/GameFramework/Controller.h:266:15: note: overridden virtual function is here
        virtual void Possess(APawn* InPawn) final; // DEPRECATED(4.22, "Posssess is marked virtual final as you should now be overriding OnPossess instead")
                     ^
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.1_of_4.cpp:3:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/ActorBlueprintFunctionLibrary.gen.cpp:8:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h:13:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/VehicleParameters.h:9:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.h:13:
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:28:18: error: constexpr variable cannot have non-literal type 'const (lambda at /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:28:25)'
  constexpr auto Cast = [](auto e) { return static_cast<typename std::underlying_type<decltype(e)>::type>(e); };
                 ^
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h:28:25: note: lambda closure types are non-literal types before C++17
  constexpr auto Cast = [](auto e) { return static_cast<typename std::underlying_type<decltype(e)>::type>(e); };
                        ^
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.4_of_4.cpp:9:
In file included from /home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/WheeledVehicleAIController.gen.cpp:8:
/home/asvnphanindra/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.h:47:8: error: declaration of 'Possess' overrides a 'final' function
  void Possess(APawn *aPawn) override;
       ^
/home/asvnphanindra/UnrealEngine/UnrealEngine/Engine/Source/Runtime/Engine/Classes/GameFramework/Controller.h:266:15: note: overridden virtual function is here
[... cut ...]
[16/22] Compile CarlaUE4GameModeBase.gen.cpp
2 warnings and 5 errors generated.
[17/22] Link (ld) libUE4Editor-OnlineSubsystemUtils.so
[18/22] Link (ld) libUE4Editor-OnlineBlueprintSupport.so
[19/22] Link (ld) libUE4Editor-OnlineSubsystemNull.so
Makefile:215: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/asvnphanindra/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2**

closed 2019-07-13 by stale[bot]

sheelabhadra, 2019-08-20:
Hi @asvnphanindra, I'm getting a similar error log with version 0.9.5. Were you able to solve the issue?