carla-simulator/carla #1585: Docker image for building Carla on Windows

nsubiron, 2019-04-29:
Make a Dockerfile that allows us building an image with a Windows 10 environment set up for building Carla.

The main complication is having Unreal built without redistributing the code/binaries.

See also #1584.

cmpute, 2019-07-07:
I guess this tool can help with dockerize UE4: https://github.com/adamrehn/ue4-docker

I didn't have enough resources to build the image so I didn't try it. But I think it can be useful for building a docker image for CARLA under windows and linux (#1584)

cmpute, 2019-07-07:
Also, make a docker image can make building on windows more easily... Definitely hope to have a docker image, and happy to help with it

germanros1987, 2020-06-26:
I would love to see this done too.. does anybody want to help?

TKeutgens, 2020-09-23:
I'm no Docker expert but I'm currently trying to build the Docker image on Windows. I managed to install all pre-requesites. Unfortunately the docker build process takes very very long. Currently I'm already waiting for almost 24 h. I feel like the process is stuck during building the UE4 Editor

Does anyone has any suggestions?

`------------------------------
Building CarlaUE4Editor...
[Upgrade]
[Upgrade] Using backward-compatible build settings. The latest version of UE4 sets the following values by default, which may require code changes:
[Upgrade]     bLegacyPublicIncludePaths = false                 => Omits subfolders from public include paths to reduce compiler command line length. (Previously: true).
[Upgrade]     ShadowVariableWarningLevel = WarningLevel.Error   => Treats shadowed variable warnings as errors. (Previously: WarningLevel.Warning).
[Upgrade]     PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs   => Set in build.cs files to enables IWYU-style PCH model. See https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/IWYU/index.html. (Previously: PCHUsageMode.UseSharedPCHs).
[Upgrade] Suppress this message by setting 'DefaultBuildSettings = BuildSettingsVersion.V2;' in CarlaUE4Editor.Target.cs, and explicitly overriding settings that differ from the new defaults.
[Upgrade]
Performing 21 actions (1 in parallel)
[1/21] Compile SharedPCH.UnrealEd.h
[2/21] Compile Module.Voice.cpp
[3/21] Link (ld) libUE4Editor-Voice.so
[4/21] Compile Module.OnlineSubsystemUtils.gen.cpp
[5/21] Compile Module.OnlineSubsystemUtils.cpp
[6/21] Link (ld) libUE4Editor-OnlineSubsystemUtils.so
[7/21] Compile Module.Carla.gen.3_of_5.cpp
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.3_of_5.cpp:2:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/OpenDrive.gen.cpp:8:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDrive.h:9:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveMap.h:10:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/road/Map.h:10:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/geom/Rtree.h:11:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry.hpp:17:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/geometry.hpp:49:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/strategies/strategies.hpp:115:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/strategies/geographic/distance_cross_track.hpp:37:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/strategies/geographic/intersection.hpp:28:
/home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/formulas/sjoberg_intersection.hpp:568:18: warning: declaration shadows a field of 'sjoberg_geodesic<CT, Order>' [-Wshadow]
        CT const asin_tj_t0j = asin(Cj * tan_betaj / sqrt_1_Cj_sqr);
                 ^
/home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/formulas/sjoberg_intersection.hpp:606:8: note: previous declaration is here
    CT asin_tj_t0j;
       ^
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.3_of_5.cpp:2:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/OpenDrive.gen.cpp:8:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDrive.h:9:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveMap.h:10:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/road/Map.h:10:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/geom/Rtree.h:11:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry.hpp:17:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/geometry.hpp:54:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/algorithms/buffer.hpp:41:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/algorithms/detail/buffer/buffer_inserter.hpp:35:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/algorithms/detail/buffer/buffered_piece_collection.hpp:28:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/algorithms/comparable_distance.hpp:23:
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/algorithms/detail/comparable_distance/implementation.hpp:22:
[... cut ...]
In file included from /home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:13:
/home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.cpp:130:11: warning: declaration shadows a field of 'FCarlaEngine' [-Wshadow]
    auto* Recorder = GetCurrentEpisode()->GetRecorder();
          ^
/home/ue4/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.h:68:19: note: previous declaration is here
  ACarlaRecorder *Recorder = nullptr;
                  ^`