carla-simulator/carla #1587: Allow custom semantic segmentation tags

nsubiron, 2019-04-29:
Our current semantic segmentation tags are [limited to 12 classes](https://carla.readthedocs.io/en/latest/cameras_and_sensors/#sensorcamerasemantic_segmentation). The system works by checking the file path of every mesh, and if the 3rd folder matches the name of one of our tags we tag it.

It would be nice if this was based on some config file (JSON?) that it's loaded at runtime, perhaps using regular expressions to match the path and filename.

germanros1987, 2020-06-26:
@marcgpuig do we have a limitation of 12 classes? My understanding is that we could have an arbitrary number of classes. Could you please confirm?

marcgpuig, 2020-07-06:
@germanros1987 no, we support a total of 255 possible classes limited by UE4. However, these classes are limited in compile time, so no more classes can be created nor modified from the PythonAPI, and I think this is what the issue is referring to.