carla-simulator/carla #1590: How to get intrinsic  and extrinsic matrix in version 0.9.3

kartik144, 2019-04-30:
Hi @marcgpuig ,
I am trying to get the intrinsic matrix in version `0.9.3` but am getting a few errors when trying your code given in this [code](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857), possibly due to version mismatch.

I have two questions:
1. What is the equavalent of `get_unreal_transform` in `0.9.3`? is it  just `get_transform`?
1. How can we access the attributes like `fov` and `image_size_x` of an RGB camera set up? I saw in the API guide that we can do it through `get_attribute` of `carla.ActorBlueprint`, but I do not understand how to get that of an actor that has already been created.

Also, it would be really great if you could tell any other changes in the code that you have given to get the extrinsic and intrinsic matrices that have to be made to use it in version `0.9.3`.
Essentially, I just want to get the intrinsic and extrinsic matrices for an RGB camera for point cloud registration.

_Originally posted by @kartik144 in https://github.com/carla-simulator/carla/issues/314#issuecomment-486956974_

ThierryDubois, 2019-05-16:
If you have any news hit me up, I'm trying to do something similar.

marcgpuig, 2019-07-17:
Hi @kartik144 
Been really busy in the new release!

> What is the equavalent of `get_unreal_transform` in `0.9.3`? is it just `get_transform`?

You have an example of Bounding Boxes available on our repo [here](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py)! 

> How can we access the attributes like `fov` and `image_size_x` of an RGB camera set up? I saw in the API guide that we can do it through `get_attribute` of `carla.ActorBlueprint`, but I do not understand how to get that of an actor that has already been created. 

Take a look at the [carla.Image](https://carla.readthedocs.io/en/latest/python_api/#carla.Image) class. We have quite improved our API documentation, hope you find it usefull!

Cheers!

gxd1994, 2019-08-21:
> I am trying to get the intrinsic matrix in version `0.9.3` but am getting a few errors when trying @marcgpuig 's code given in this [gist](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857), possibly due to version mismatch.
> 
> I have two questions:
> 
> 1. What is the equavalent of `get_unreal_transform` in `0.9.3`? is it  just `get_transform`?
> 2. How can we access the attributes like `fov` and `image_size_x` of an RGB camera set up? I saw in the API guide that we can do it through `get_attribute` of `carla.ActorBlueprint`, but I do not understand how to get that of an actor that has already been created.
> 
> Also, it would be really great if someone could tell any other changes in the [code](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857) that have to be done to get the extrinsic and intrinsic matrices in version `0.9.3`.
> 
> Essentially, I just want to get the intrinsic and extrinsic matrices for an RGB camera for point cloud registration. So if there is any other easier way to get them, that would also be helpful.
> 
> _Originally posted by @kartik144 in [#314 (comment)](https://github.com/carla-simulator/carla/issues/314#issuecomment-486956974)_

i have same problem, i can get correct extrinsic in version `0.9.5`

gxd1994, 2019-08-21:
> I am trying to get the intrinsic matrix in version `0.9.3` but am getting a few errors when trying @marcgpuig 's code given in this [...]

Have you solved this problem yet?  can you help me, thx

noahcao, 2019-08-31:
I'm using 0.9.4. How could I get the focal loss of the rgb camera? Thx.

gnaneshkm, 2019-09-11:
> Have you solved this problem yet? can you help me, thx

> i have same problem, i can get correct extrinsic in version `0.9.5`

gxd1994 can you please explain me how you get extrinsic parameters  in 0.9.5. thanks in advance.

closed 2019-11-25 by stale[bot]

marcgpuig, 2020-10-15:
I've added a new script called `lidar_to_camera.py` under `examples` folder with an example on how to project 3D lidar points to an RGB camera, including how to get the K (intrinsic) matrix. This example can also be used for other 3D points at your like.
Check out #3454.

Cheers!