carla-simulator/carla #1595: Autopilot code location?

Abanoub-G, 2019-04-30:
Hi,
I am looking for the hard coded AI that is used by the non player characters. Is it located in /carla/Unreal/CalraUE4/Plugins/Carla/Source/Carla/Vehicle  ?
If yes, then which file is it from all of the ones in there?

I want to modify the autopilot to make it follow the RoutePlanners to reach a destination point. To do so I need to understand how your autopilot works and in particular I am trying to reach the point in the autopilot code where it randomly decides which route to take in junctions. So it would be much appreciated if you point me towards the files that I am looking for. 

Thanks

lijecaru, 2019-05-01:
It sounds to me that you could just modify the 'basic agent' associated files to perform your task. In this case, you would need to go to the file 'automatic_control.py' and see that 'BasicAgent' is selected as default. Then go to the PythonAPI/agents/navigation/basic_agent.py and modify the function `def set_destination(self, location)` in the option` end_waypoint`. You can get your desired final waypoint from the map itself. 
The vehicle automatically follows the planned routes, it computes the next waypoint at every timestep, and every waypoint is located in a planned route.

Abanoub-G, 2019-05-07:
Hi @lijecaru  

Thanks a lot for your reply. Yes indeed that is what I was looking for. How do I set this end_waypoint from another .py script. 

For example in the spawn_npc.py example it spawns the vehicles but I can't find where does it set the vehicles' destinations to. How would I be able to do that? Would it be something like set_destination(self, carla.location(x,y,z))

Thanks

Abanoub-G, 2019-05-07:
Also what is the diffference between the autopilot in spawn_npc.py and the one in automatic_control.py/basic_agent.py ?

Thanks

Derekabc, 2019-05-07:
I am not sure.  The autopilot in spawn_npc.py is implemented by the server side and following the route generated by opendrive, and it looks smoothly when running.
The one in automatic_control.py is controlled by client-side and PID controller.  It performs worsely i think.

Abanoub-G, 2019-05-07:
Hi @Derekabc 
Yes, I totally agree. The autopilot in spawn_npc.py follows the splines and performs really well. The one in automatic_control.py is pretty awful.

Any idea where is the code for the autopilot used in spawn_npc.py i.e. the one that follows the splines?

Thanks

Abanoub-G, 2019-05-07:
Thanks @Derekabc,  but that doesn't show any of the AI used or how it follows the way points in the route planner...?
What does the file /carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp  do then?

morningsky, 2019-05-18:
> Also what is the diffference between the autopilot in spawn_npc.py and the one in automatic_control.py/basic_agent.py ?
> 
> Thanks

Do you know the difference now? I have the same question...

closed 2019-07-24 by stale[bot]

SamiaAbid, 2019-12-26:
I have the issue, can anybody please guide me how to assign a specific destination to the vehicle? where it must stop?

Derekabc, 2019-12-26:
@SamiaAbid  You can check the location of the vehicle and set the control.steer = 0, control.throttle = 0 and control.brake = -1 if the location meets the requirement.