carla-simulator/carla #1597: Python API documentation

adarsh9975, 2019-05-01:


nsubiron, 2019-05-02:
We're working on it, but with the challenge ahead we're always having something with higher priority :sweat: 

In the meantime, I would recommend taking a look at the source code

- In the C++ API most of the methods are commented, almost for every Python function there's the equivalent in C++, at [LibCarla/source/carla/client](https://github.com/carla-simulator/carla/tree/master/LibCarla/source/carla/client).
- Then to check if a function is really there in Python you can take a look at the bindings at [PythonAPI/carla/source/libcarla](https://github.com/carla-simulator/carla/tree/master/PythonAPI/carla/source/libcarla)

Hope it helps!

adarsh9975, 2019-05-02:
Thanks a lot 😊. I'll begin tinkering with this.

closed 2019-05-02 by adarsh9975