carla-simulator/carla #1599: There is a delay in spawning.

layssi, 2019-05-01:
Example code:
vehicle = world.try_spawn_actor(vehicle_blueprint, transform)
world_actor_list = world.get_actors()
world_actor_list = world_actor_list.filter("*vehicle*")
print(len(world_actor_list))

In the print statement above, the number of actors "might" increase or might not (even when vehicle is properly spawned). The number of vehicles will increase if you get world actors after some delay. so there is some delay when you spawn. 

This is not a big issue that needs fixing. I am just not sure if I should be expecting a delay in every response. I have been seeing situations where I get proper response but the world is not updated.

Any info would be helpful.

nsubiron, 2019-05-02:
This happens because `get_actors()` retrieves a cached list of actor as received in the last "world tick", I explained a bit more in detail here https://github.com/carla-simulator/carla/issues/1274#issuecomment-465567495.

You can force your script to wait for a "world tick", this way you'll see your newly spawned actor in the list

```py
vehicle = world.spawn_actor(bp, transform)
world.wait_for_tick()
actors = world.get_actors()
```

Marwa215, 2019-05-02:
Hi, @nsubiron I have a problem in spawning actors through configuration xml file in scenario runner carla9.5 for town05, the problem if I selected any position for spawning the vehicle it gives me error due to a collision at this location even if I tried to select random location from the given spawning list it gives me error for collision at the spawn point (0, 0, 0). Why I face such strange problem ....any help will be appreciated

nsubiron, 2019-05-03:
Hi @Marwa215, I'm not very familiar with scenario runner, you may try opening an issue in their repository.

closed 2019-07-09 by stale[bot]