carla-simulator/carla #1600: Can't control in manual_control.py [0.9.5]

ChrisHRZ, 2019-05-01:
When I run the client of manual_control.py. I can't control my car with w, s, a and d. And other button like h for help or c for change the weather didn't work either. I can only control my car with up, down, left and right. Is this a bug in carla 0.9.5? How can i fix this problem? Thank you!

JMorceaux, 2019-05-02:
We're gonna need a few more infos about your problem:
* can you show us your console ?
* What is the version of pygame ?
* What type of keyboard are you using (qwerty , azerty, etc ...) ?

ChrisHRZ, 2019-05-11:
> We're gonna need a few more infos about your problem:
> 
> * can you show us your console ?
> * What is the version of pygame ?
> * What type of keyboard are you using (qwerty , azerty, etc ...) ?

I found that I can control my vehicle with ctrl+w,s,a and d. And other letter key only work with ctrl+this letter key. And answer your question:

- I build Carla on ubuntu 18.04
- My pygame version is 1.9.6
-I use qwerty keyboard

Thank you again.

closed 2019-07-17 by stale[bot]