carla-simulator/carla #1605: Replaying without the ego vehicle

ilikecyf86, 2019-05-03:
Hi CARLA team:
I'm using CARLA to test route planning algorithm. The logging function is very useful for identifying problems of the planner. However, sometimes when there is a collision, I might want to modify the algorithm and test it in the exact same scenario again. The current replay function doesn't allow me to do that. It would be nice if I can replay everything except the ego vehicle, and do the planning again.

Currently, I'm thinking about removing everything about the ego vehicle in the log file and replay it. Is there any document about the format of the log file that I can refer to? Or it would be nice if you can add this feature into CARLA.

germanros1987, 2020-06-26:
@Axel1092 does this look like something you could add in a short amount of time?

bernatx, 2020-07-01:
Hi, that feature was added time ago, in CARLA 0.9.8.
You can use the flag -i that means that all hero vehicle will not be simulated in the replay.

closed 2020-07-01 by bernatx